realsenseai/librealsense #5181: Frame didn't arrived within 5000 - try_fetch_usb_device failed using D435 - T265 recognized

RodolfoAndrade, 2019-11-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 / D435 } | 
| Firmware Version                   | {5.11.15.0} | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.29.0 }                          | 
| Language                            |  {C++/python }                          | 
| Segment			|  {others }                          | 

### Issue Description
We are trying to run a simple realsense example for D435 and also for D415. The exception message says that "Frame didn't arrived within 5000".

The Intel RealSense Viewer works fine.

We need to run the code using python and visual studio. We tried to build from source and using the pip install pyrealsense2.

There is no issue similar to our error because we have the log message saying something about T265 while we are using D435.

Here is the console output from this line of code "rs.log_to_file(rs.log_severity.debug, file_path='./log.txt')":
 04/11 16:53:26,750 DEBUG [7500] (ds5-factory.cpp:881) try_fetch_usb_device(...) failed.
 04/11 16:53:26,750 DEBUG [7500] (tm-context.cpp:67) T265 query acomplished after 0 ms]
 04/11 16:53:26,766 INFO [7500] (environment.cpp:85) Found 299 unreachable streams, 299 extrinsics deleted
 04/11 16:53:26,917 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,917 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,917 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,917 INFO [7500] (sensor.cpp:367) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
 04/11 16:53:26,931 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,931 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,931 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,931 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,991 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,991 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:26,991 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:27,011 DEBUG [7500] (global_timestamp_reader.cpp:109) time_diff_keeper::start: _users_count = 1
 04/11 16:53:27,023 DEBUG [7500] (global_timestamp_reader.cpp:109) time_diff_keeper::start: _users_count = 2
 04/11 16:53:35,067 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:35,067 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:35,067 INFO [7500] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 04/11 16:53:35,071 DEBUG [7500] (global_timestamp_reader.cpp:120) time_diff_keeper::stop: _users_count = 1
 04/11 16:53:35,579 DEBUG [7500] (global_timestamp_reader.cpp:120) time_diff_keeper::stop: _users_count = 0
 04/11 16:53:35,579 DEBUG [7500] (global_timestamp_reader.cpp:123) time_diff_keeper::stop: stop object.
 04/11 16:53:35,579 DEBUG [22312] (global_timestamp_reader.cpp:183) Notification: time_diff_keeper polling loop is being shut-down

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

@RodolfoAndrade Could you please provide the code snippet that you get the issue so that we can further support on this? Thanks!

RodolfoAndrade, 2019-11-27:
This is the python script we were trying to run.

    import pyrealsense2 as rs
    try:
        # Create a context object. This object owns the handles to all connected realsense devices
        pipeline = rs.pipeline()
        pipeline.start()
        rs.log_to_file(rs.log_severity.debug, file_path='./log.txt')
        
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        
        exit(0)
        
    except Exception as e:
        print(e)

Thanks for your reply!

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@RodolfoAndrade With your code snippet we couldn't reproduce the issue. Did you try other python examples in SDK package? Did you have the same issue?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@RodolfoAndrade Could you please update? Thanks!

RodolfoAndrade, 2019-12-19:
Currently I am trying the rs.align cpp example code on visual studio 2017.
```
// Create a pipeline to easily configure and start the camera
 rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_stream(RS2_STREAM_COLOR);
pipe.start(cfg);
rs2::log_to_console(rs2_log_severity::RS2_LOG_SEVERITY_DEBUG);
```

This is the log printed to the console:
```
 19/12 18:12:55,231 INFO [20056] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 19/12 18:12:55,233 INFO [20056] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 19/12 18:12:55,234 INFO [20056] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 19/12 18:12:55,238 DEBUG [20056] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 1
 19/12 18:12:55,770 DEBUG [20056] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 0
 19/12 18:12:55,770 DEBUG [20056] (global_timestamp_reader.cpp:146) time_diff_keeper::stop: stop object.
 19/12 18:12:55,771 DEBUG [11004] (global_timestamp_reader.cpp:206) Notification: time_diff_keeper polling loop is being shut-down
RealSense error calling rs2_pipeline_wait_for_frames(pipe:000001E6F9127300):
    Frame didn't arrived within 15000

C:\Users\Santos\Downloads\realsense\librealsense-master-2.31.0\build\Debug\rs-align.exe (process 18544) exited with code 1.
```

We have updated the github repository to the 2.31.0 version and followed the compiling instruction from realsense to generate the visual studio project.

We tried another example from the rs-color C code, here is the output:
```
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 828112074018
    Firmware version: 05.11.15.00

rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:00000248965C0950):
    Frame didn't arrived within 15000

C:\Users\Santos\Downloads\realsense\librealsense-master-2.31.0\build\Debug\rs-color.exe (process 16928) exited with code 1.
```

Sorry for the delay. Thank you for the reply!

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@RodolfoAndrade Could you please try to configure RS2_STREAM_DEPTH and RS2_STREAM_COLOR to certain supported resolution such as 640x480 to try rs-align or rs-capture again?

And did you get such issue if you run SDK examples(such as rs-align) from SDK installation package? https://github.com/IntelRealSense/librealsense/releases/download/v2.32.1/Intel.RealSense.SDK-WIN10-2.32.1.1297.exe

RodolfoAndrade, 2020-02-07:
We downloaded and install the new version (2.32.1.1297). The Intel RealSense Viewer works fine and the rs-align.exe found in "C:\Program Files (x86)\Intel RealSense SDK 2.0\tools" works too.

But the same error occurs with the VisualStudio project on the sample folder:
```
 07/02 15:15:23,755 INFO [14328] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 07/02 15:15:23,775 INFO [14328] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 07/02 15:15:23,899 INFO [14328] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 07/02 15:15:24,096 DEBUG [14328] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 1
 07/02 15:15:24,760 DEBUG [14328] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 0
 07/02 15:15:24,775 DEBUG [14328] (global_timestamp_reader.cpp:146) time_diff_keeper::stop: stop object.
RealSense error calling rs2_pipeline_wait_for_frames(pipe:000002414A0B5DA0):
    Frame didn't arrive within 15000
```

We also try your suggestion to change the resolution. Here is the code:
```
    // Create and initialize GUI related objects
    window app(640, 480, "RealSense Align Example"); // Simple window handling
    ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
    rs2::colorizer c;                     // Helper to colorize depth images
    texture depth_image, color_image;     // Helpers for renderig images

    // Create a pipeline to easily configure and start the camera
    rs2::pipeline pipe;
    rs2::config cfg;
    //cfg.enable_stream(RS2_STREAM_DEPTH);
    //cfg.enable_stream(RS2_STREAM_COLOR);
    cfg.enable_stream(RS2_STREAM_DEPTH, -1, 640, 480, rs2_format::RS2_FORMAT_RGB8, 0);
    cfg.enable_stream(RS2_STREAM_COLOR, -1, 640, 480, rs2_format::RS2_FORMAT_RGB8, 0);
    pipe.start(cfg);
    rs2::log_to_console(rs2_log_severity::RS2_LOG_SEVERITY_DEBUG);
```
 This was the error message:
```
RealSense error calling rs2_pipeline_start_with_config(pipe:000002C8E40B71F0, config:000002C8E40B72B0):
    Couldn't resolve requests
```

Hope this was helpful. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@RodolfoAndrade  For depth, please modify the format from RS2_FORMAT_RGB8 to RS2_FORMAT_Z16 to try again?

    cfg.enable_stream(RS2_STREAM_DEPTH, -1, 640, 480, rs2_format::RS2_FORMAT_RGB8, 0);

RodolfoAndrade, 2020-02-11:
I did the modification:
`cfg.enable_stream(RS2_STREAM_DEPTH, -1, 640, 480, rs2_format::RS2_FORMAT_Z16, 0);`
but it did not work. This was the output:
```
 11/02 14:43:52,249 INFO [15692] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 11/02 14:43:52,255 INFO [15692] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 11/02 14:43:52,259 INFO [15692] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 11/02 14:43:52,265 DEBUG [15692] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 1
 11/02 14:43:52,444 DEBUG [15692] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 0
 11/02 14:43:52,447 DEBUG [15692] (global_timestamp_reader.cpp:146) time_diff_keeper::stop: stop object.
 11/02 14:43:52,450 DEBUG [16232] (global_timestamp_reader.cpp:206) Notification: time_diff_keeper polling loop is being shut-down
RealSense error calling rs2_pipeline_wait_for_frames(pipe:00000248F70036C0):
    Frame didn't arrive within 15000
```

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@RodolfoAndrade Could you please try if rs-enumerate can run successfully for your own build? And please also provide the output if it can run successfully. Thanks!

RodolfoAndrade, 2020-02-19:
This was the output for rs-enumerate-devices:
```
Device info:
    Name                          :     Intel RealSense D435
    Serial Number                 :     828112074018
    Firmware Version              :     05.12.02.100
    Recommended Firmware Version  :     05.12.02.100
    Physical Port                 :     \\?\usb#vid_8086&pid_0b07&mi_00#6&5e223c0&0&0000#{24e552d7-6523-47f7-a647-d3465bf1f5ca}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B07
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     828313021199
    Firmware Update Id            :     828313021199

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format
    Infrared 2    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 15Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y8
    Infrared 2    1280x800      @ 15Hz     Y8
    Infrared 2    1280x720      @ 30Hz     Y8
    Infrared 1    1280x720      @ 30Hz     Y8
    Infrared 1    1280x720      @ 15Hz     Y8
    Infrared 2    1280x720      @ 15Hz     Y8
    Infrared 2    1280x720      @ 6Hz      Y8
    Infrared 1    1280x720      @ 6Hz      Y8
    Infrared 2    848x480       @ 90Hz     Y8
    Infrared 1    848x480       @ 90Hz     Y8
    Infrared 1    848x480       @ 60Hz     Y8
    Infrared 2    848x480       @ 60Hz     Y8
    Infrared 2    848x480       @ 30Hz     Y8
    Infrared 1    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 15Hz     Y8
    Infrared 1    848x480       @ 15Hz     Y8
    Infrared 2    848x480       @ 6Hz      Y8
    Infrared 1    848x480       @ 6Hz      Y8
    Infrared 2    848x100       @ 100Hz    Y8
    Infrared 1    848x100       @ 100Hz    Y8
    Infrared 2    640x480       @ 90Hz     Y8
    Infrared 1    640x480       @ 90Hz     Y8
    Infrared 2    640x480       @ 60Hz     Y8
    Infrared 1    640x480       @ 60Hz     Y8
    Infrared 1    640x480       @ 30Hz     Y8
    Infrared 2    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 6Hz      Y8
    Infrared 1    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 25Hz     Y16
    Infrared 1    640x400       @ 15Hz     Y16
    Infrared 2    640x400       @ 15Hz     Y16
    Infrared 2    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 15Hz     Y8
    Infrared 2    640x360       @ 15Hz     Y8
    Infrared 2    640x360       @ 6Hz      Y8
    Infrared 1    640x360       @ 6Hz      Y8
[... 232 lines cut ...]
    Color         320x180       @ 60Hz     Y16
    Color         320x180       @ 60Hz     BGRA8
    Color         320x180       @ 60Hz     RGBA8
    Color         320x180       @ 60Hz     BGR8
    Color         320x180       @ 60Hz     YUYV
    Color         320x180       @ 30Hz     RGB8
    Color         320x180       @ 30Hz     Y16
    Color         320x180       @ 30Hz     BGRA8
    Color         320x180       @ 30Hz     RGBA8
    Color         320x180       @ 30Hz     BGR8
    Color         320x180       @ 30Hz     YUYV
    Color         320x180       @ 6Hz      RGB8
    Color         320x180       @ 6Hz      Y16
    Color         320x180       @ 6Hz      BGRA8
    Color         320x180       @ 6Hz      RGBA8
    Color         320x180       @ 6Hz      BGR8
    Color         320x180       @ 6Hz      YUYV

C:\Users\Santos\Downloads\realsense\librealsense\build\Debug\rs-enumerate-devices.exe (process 14116) exited with code 0.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .

```

I hope it helps. Thanks!

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

@RodolfoAndrade I didn't see any abnormal from the output of rs-enumerate-devices. Do you have both python2 and python3 installed on your system? If so, could you please try to force compilation with a specific version with **-DPYTHON_EXECUTABLE=<path to python executable>** and try again? Thanks!

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@RodolfoAndrade Any other questions about this? Thanks!

RodolfoAndrade, 2020-03-06:
We are trying to use the Visual Studio instead of python, at the moment. But both give the same error.

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@RodolfoAndrade Sorry that I couldn't reproduce the issue here. I understood pre-built RealSenseViewer in SDK package has no problem on your setup but build from source has problem. Considering this, could you please double check if you follow the build procedure as  described in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md ?

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@RodolfoAndrad Sorry that we didn't hear from you for weeks, will close this issue at this point. Please create another new one if you still have issues or questions. Thanks!

closed 2020-04-22 by RealSenseCustomerSupport