realsenseai/librealsense #5182: T265: How to Set Exposure Mode Before Starting the Pipeline

JBBee, 2019-11-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.2.0.857 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                 | 
| Platform                                 | PC |
| SDK Version                            |  2.30.0                         | 
| Language                            | C#                   | 
| Segment			|  AR                         | 

### Issue Description
I have a question about how to set an option on a sensor when using the pipeline interface prior to starting the pipeline.  For example the following code throws the exception: `To control exposure you must set sensor to manual exposure mode prior to streaming`  on the line `ps2.Options[Option.Exposure].Value = 16000;`  How do I set the sensor to manual exposure mode before starting the pipeline?

       static void Main(string[] args)
        {
            using (var ctx = new Context())
            {
                var devices = ctx.QueryDevices();
                if (devices.Count == 0) return;
                var dev = devices[0]; //T265

                var poseSensor = dev.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
                var ps = PoseSensor.FromSensor(poseSensor);
                ps.Options[Option.EnableAutoExposure].Value = 0;

                var pipeline = new Pipeline(ctx);
                var pp = pipeline.Start(FC);

                var poseSensor2 = pp.Device.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
                var ps2 = PoseSensor.FromSensor(poseSensor2);

                ps2.Options[Option.Exposure].Value = 16000;
                ps2.Options[Option.Gain].Value = 1.4f;

                AutoResetEvent stopEvent = new AutoResetEvent(false);
                Console.CancelKeyPress += (s, e) =>
                {
                    e.Cancel = true;
                    stopEvent.Set();
                };
                stopEvent.WaitOne();

                pipeline.Stop();
                Thread.Sleep(1000);
            }
            System.Console.WriteLine("Done");
        }

Thanks.

MartyG-RealSense, 2019-11-04:
SDK version 2.230.0 in June 2019 added a manual exposure option for the T265's fisheye sensor.  The instructions for how to configure it are in the link below.

https://github.com/IntelRealSense/librealsense/pull/3992

JBBee, 2019-11-05:
Thanks @MartyG-RealSense , that answers my question regarding how to use the manual exposure.  But I'm still not clear on how to set an option on a sensor before the pipeline starts.  For example, I added the enums to the C# wrapper for the newer T265 options, which are required to be set before the sensor is started.  But the following code prints: `Pose jumping enabled: 1` when I would like to disable pose jumping.  How can I accomplish this while still using the pipeline interface?

        static void Main(string[] args)
        {
            using (var ctx = new Context())
            {
                var devices = ctx.QueryDevices();
                if (devices.Count == 0) return;
                var dev = devices[0]; //T265

                var poseSensor = dev.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
                var ps = PoseSensor.FromSensor(poseSensor);

                ps.Options[Option.EnablePoseJumping].Value = 0;

                var pipeline = new Pipeline(ctx);
                var pp = pipeline.Start(f => { using (f) { Console.WriteLine("Got Frame"); } });

                var poseSensor2 = pp.Device.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
                var ps2 = PoseSensor.FromSensor(poseSensor2);

                System.Console.WriteLine("Pose jumping enabled: " + ps2.Options[Option.EnablePoseJumping].Value);
                
                AutoResetEvent stopEvent = new AutoResetEvent(false);
                Console.CancelKeyPress += (s, e) =>
                {
                    e.Cancel = true;
                    stopEvent.Set();
                };
                stopEvent.WaitOne();

                pipeline.Stop();
                Thread.Sleep(1000);
            }
            System.Console.WriteLine("Done");
        }

Thanks.

MartyG-RealSense, 2019-11-05:
Pose jumping can be turned off by setting RS2_OPTION_ENABLE_POSE_JUMPING to false.

https://github.com/IntelRealSense/librealsense/commit/de76d3a1bdd094100a3985e4dd3273e8e1fd2bb6

JBBee, 2019-11-06:
Thanks @MartyG-RealSense , I do disable it which is this line in the code segment above: `ps.Options[Option.EnablePoseJumping].Value = 0;`  But once I start the pipeline it shows as re-enabled, if you take a look at the code above.

MartyG-RealSense, 2019-11-06:
Reading through the code, it gives me the impression that it is setting the variable **ps.Options[Option.EnablePoseJumping].Value** to '0' and then printing out the value (0) stored in that variable, rather than printing the actual current status of the pose jumping function.  If so, this would give an incorrect impression that pose jumping is disabled when it is actually still enabled.  What do you think?

JBBee, 2019-11-08:
@MartyG-RealSense , the code above is using the Intel.Realsense C# wrapper so `ps.Options[Option.EnablePoseJumping].Value = 0;` is setting the pose jumping function and `ps2.Options[Option.EnablePoseJumping].Value` is reading the current value after starting the pipeline.  The code above sets it to zero before starting the pipeline, then once the pipeline is started it reads back a 1.  So my question is how do I set it to zero and have it persist after starting the pipeline?

Thanks.

MartyG-RealSense, 2019-11-08:
I will have to defer this question to one of the Intel team such as @dorodnic since my coding knowledge is insufficient to provide a solution to this particular question.

dorodnic, 2019-11-11:
Hi @JBBee 
Sorry, this may be a bit confusing. The way to solve this is by using `Config` object, something like this:
```
using (var ctx = new Context())
        {
            var config = new Config();
            var pipeline = new Pipeline();
            var profile = config.Resolve(pipeline);
            var dev = profile.Device;
            var poseSensor = dev.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
            var ps = PoseSensor.FromSensor(poseSensor);

            ps.Options[Option.EnablePoseJumping].Value = 0;

            var pp = pipeline.Start(config, f => { using (f) { Console.WriteLine("Got Frame"); } });

            var poseSensor2 = pp.Device.QuerySensors<Sensor>().First(s => s.Is(Extension.PoseSensor));
            var ps2 = PoseSensor.FromSensor(poseSensor2);

            System.Console.WriteLine("Pose jumping enabled: " + ps2.Options[Option.EnablePoseJumping].Value);
            
            AutoResetEvent stopEvent = new AutoResetEvent(false);
            Console.CancelKeyPress += (s, e) =>
            {
                e.Cancel = true;
                stopEvent.Set();
            };
            stopEvent.WaitOne();

            pipeline.Stop();
            Thread.Sleep(1000);
        }
        System.Console.WriteLine("Done");
    }
```

JBBee, 2019-11-11:
Ahh, thank you.  This worked great @dorodnic @MartyG-RealSense

closed 2019-11-11 by JBBee

zainmehdi, 2019-12-06:
@dorodnic I am unable to set Relocalization and pose jumping option using C++ interface. If I use pipeline I get the error " Option is read-only while streaming" and If I try to access device using enumeration on raspberry pi 4 it doesnt return any device even when its connected. Can you please specify a way ? Thanks

zainmehdi, 2019-12-06:
for anyone trying stuck at the same problem I was here is the way to do it in C++ following @dorodnic approach for C#

`rs2::config cfg;
rs2::pipeline pipe;
rs2::device dev;
rs2::sensor sensor;
auto profile = cfg.resolve(pipe);
dev=profile.get_device();
sensor=dev.query_sensors().front();
sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING,0);`