realsenseai/librealsense #5183: Getting raw depth values that are aligned to color

aalarcon96, 2019-11-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

aalarcon96, 2019-11-04:
[image]
[image]

As you can see, even after using the align-advanced sample, there seems to be a misalignment. You can see that there is a bigger space between my thumb and the edge of the screen on the color image versus the depth. Is there a way to properly align them and get the depth values?

ev-mp, 2019-11-05:
@aalarcon96 hello, at closer examination it is possible that you misinterpret the data due to the blend of the overlayed depth with the background RGB image, and do not account for the blank lines at the bottom of depth image, which are integral part of if (the black strip).
[image]

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

@aalarcon96 Any luck to get it through with the suggestion of ev-mp? Looking forward to your update. Thanks!

aalarcon96, 2019-11-21:
Ev-mp was right, thank you for following up.

closed 2019-11-21 by aalarcon96