realsenseai/librealsense #5184: T265: A question regarding the rs-pose-apriltag example

neilyoung, 2019-11-04:
I'm trying to prove my understanding of this sample: The code outputs two extrinsic transformation matrices: One for the camera position at the moment of the detection and - if there is good confidence - also some other extrinsic transformation matrix, called "world". As far as I understand that, this is the position of the AprilTag in the coord system of the camera. 

If this is the case, is my assumption correct, that the pose of the AprilTag (if it doesn't physically move) MUST NOT change significantly while I'm roaming around the tag?

neilyoung, 2019-11-04:
I think my assumption is correct :)

honpong, 2019-11-06:
@neilyoung, at line https://github.com/IntelRealSense/librealsense/blob/9f99fa9a509555f85bffc15ce27531aaa6db6f7e/examples/pose-apriltag/rs-pose-apriltag.cpp#L214

`tags.pose_in_camera[t]` is the tag position and orientation in camera coordinate.
`tags.pose_in_world[t]` is the tag position and orientation in world coordinate which is the same as the camera pose coordinate. 
This example does **not** output camera position nor orientation.

neilyoung, 2019-11-06:
Sorry,  but IMHO you are printing two lines: camera (rotation matrix and translation vector) and "world" (also rotation and translation). Both are different. From my understanding the "camera" is the pose of the camera in camera coordinates and "world" the the pose of the tag. Is this what you mean? Then we are on the same page.

honpong, 2019-11-06:
No, “camera” is the tag pose in “camera” coordinate and “world” is the tag pose in “world” coordinate. Both are pose of the same tag, just in different coordinate systems.

neilyoung, 2019-11-06:
OK, I think I got it, with the exception of the "world" coordinate system. What exactly is that? I have heard/read this term in several samples/publications already, w/o having a really good idea, what that is. Especially if you say (a couple of posts above):

> tags.pose_in_world[t] is the tag position and orientation in world coordinate which is the same as the camera pose coordinate.

If it is the same as the camera pose coordinate, why it is different :)?

Sorry for being so hard-minded. Please bear with me.

honpong, 2019-11-06:
@neilyoung “world” is a name of a coordinate system. I may call it system1, system2, etc... Using the word “world” just means it is the system of the highest level known in the problem.

neilyoung, 2019-11-06:
Ah great. Thanks for the explanation. Didn’t expect this meaning of the word. Do you know about any good source of information for continued learning?

Thank you very much.

neilyoung, 2019-11-07:
Can I assume, that the `tag.pose_in_camera` translation vector meets the center of the tag? I still have some problems with the rotation matrix content of the `tag.pose_in_camera`, the `yaw` value makes not much sense to me, need to double check. But in case the translation vector describes the center of the tag, I could live with the `yaw` of the initial camera pose.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @neilyoung
Is there anything else necessary on this thread?

neilyoung, 2020-01-27:
Nope

closed 2020-01-27 by neilyoung