realsenseai/librealsense #5185: Convert Depth frame to OpenCV Mat (CV_16UC1) in android Java.

kafan1986, 2019-11-04:
I am building an App on Android and need some help. I have and I am looking at the examples provided but there are some queries.

How to convert a depth frame to opencv Mat of type CV_16UC1 and a colour image (BGR) to  CV_8UC3 in java? Or should I write in native and invoke via JNI? I know it can be done in c++ via below code:

```
// For Depth Frame
rs2::frame depth_frame = frames.get_depth_frame();
Mat ir(Size(640, 480), CV_16UC1, (void*)depth_frame.get_data(), Mat::AUTO_STEP);

// For Color Frame
rs2::frame color_frame = frames.get_color_frame();
Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
```

kafan1986, 2019-11-06:
> I am building an App on Android and need some help. I have and I am looking at the examples provided but there are some queries.
>
> How to convert a depth frame to opencv Mat of type CV_16UC1 and a [...]

I found the answer.

DepthFrame depth = f.as(Extension.DEPTH_FRAME);
Mat testMap = new Mat(depth.getHeight(),depth.getWidth(),CV_16UC1);
int size = (int) (testMap.total() * testMap.elemSize());
byte[] return_buff = new byte[size];
depth.getData(return_buff);
short[] shorts = new short[size/2];
ByteBuffer.wrap(return_buff).order(ByteOrder.LITTLE_ENDIAN).asShortBuffer().get(shorts);
testMap.put(0, 0, shorts);
// You check a pixel value by using soomething like this (double)testMap.get(depth.getHeight()/2,depth.getWidth()/2)[0];

closed 2019-11-06 by kafan1986

vshalkov, 2019-12-13:
@kafan1986  Can you help me?
How i can save depth map to image file (.tiff or .png)?
if a use your code:
```
DepthFrame depth = f.as(Extension.DEPTH_FRAME);
Mat testMap = new Mat(depth.getHeight(),depth.getWidth(),CV_16UC1);
int size = (int) (testMap.total() * testMap.elemSize());
byte[] return_buff = new byte[size];
depth.getData(return_buff);
short[] shorts = new short[size/2];
ByteBuffer.wrap(return_buff).order(ByteOrder.LITTLE_ENDIAN).asShortBuffer().get(shorts);
testMap.put(0, 0, shorts);
```
And i try to convert to Bitmap:
```
val bmp = Bitmap.createBitmap(testMap.cols(), testMap.rows(), Bitmap.Config.ARGB_8888)
Utils.matToBitmap(testMap, bmp) <--------- this code freeze android app
```
My android app freezes.

I asked a question here:
https://github.com/IntelRealSense/librealsense/issues/5446
Could you help me?
Thanks!

kafan1986, 2019-12-14:
> @kafan1986 Can you help me?
> How i can save depth map to image file (.tiff or .png)?
> if a use your code:
>
> ```
> DepthFrame depth = f.as(Extension.DEPTH_FRAME);
> Mat testMap = new [...]

Already answered on your original question thread.