realsenseai/librealsense #5186: Frame didn't arrived within 15000

sumandeepb, 2019-11-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 (D435i) } | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Linux (Ubuntu 18.04.3 LTS)  | 
| Kernel Version (Linux Only)    |  5.0.0-32                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.29.0                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
Error on trying to get depth frames from RealSense D435i. I have 8 devices (not connected simultaneously, same code being tested on devices one by one). This happens only on 2 out of 8.

Error 1:
Detected device: 846112072511
terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 15000
Aborted (core dumped)

Error 2 (using catch exception):
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)

Error 3 (using catch rs2::error):
error type: UNKNOWN
failed_function: rs2_pipeline_wait_for_frames
failed args: pipe:0x563588aa1370

Output for $ rs-fw-update -l
connected devices:
1) Name: Intel RealSense D435I, serial number: 846112072511, update serial number: 904523051334, firmware version: 05.11.06.250, USB type: 3.2

Output for $ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	846112072511
    Firmware Version              : 	05.11.06.250
    Recommended Firmware Version  : 	05.11.15.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	904523051334
    Firmware Update Id            : 	904523051334

sumandeepb, 2019-11-05:
Upgrading to firmware 05.11.15.00 did not cause any change.

sumandeepb, 2019-11-05:
`// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

// include OpenCV header file
#include <opencv2/opencv.hpp>

#include <string>
#include <sstream>
#include <sys/stat.h>

const int WIDTH = 848;
const int HEIGHT = 480;
const int FPS = 30;

int main(int argc, char *argv[])
{
    rs2::context ctx; // Create librealsense context for managing devices
    std::vector<rs2::pipeline> pipelines;
    std::vector<std::string> serialNos;
    std::vector<cv::Mat> depth16_frames;
    std::vector<cv::Mat> depth8_frames;
    std::vector<cv::Mat> ifred_frames;

    // Start a streaming pipe per each connected device
    for (auto &&dev : ctx.query_devices())
    {
        try
        {
            //Contruct a pipeline which abstracts the device
            rs2::pipeline pipe;

            //Create a configuration for configuring the pipeline with a non default profile
            rs2::config cfg;

            //Enable specific device
            cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

            //Add desired streams to configuration
            //cfg.enable_stream(RS2_STREAM_COLOR, WIDTH, HEIGHT, RS2_FORMAT_BGR8, FPS);
            cfg.enable_stream(RS2_STREAM_DEPTH, WIDTH, HEIGHT, RS2_FORMAT_Z16, FPS);
            cfg.enable_stream(RS2_STREAM_INFRARED, WIDTH, HEIGHT, RS2_FORMAT_Y8, FPS);

            //Instruct pipeline to start streaming with the requested configuration
            pipe.start(cfg);

            // add pipe and device serial id to list
            pipelines.emplace_back(pipe);
            serialNos.emplace_back(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
            depth16_frames.emplace_back(cv::Mat());
            depth8_frames.emplace_back(cv::Mat(cv::Size(WIDTH, HEIGHT), CV_8UC1));
            ifred_frames.emplace_back(cv::Mat());

            std::cout << "Detected device: " << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
        }
        catch (...)
        {
            std::cout << "Exception Occured" << std::endl;
        }
    }

    int num_devices = pipelines.size();
    std::time_t sys_time;
    std::string filePath("/home/sumandeepb/ExtDrv/data/realsense/filter/");
    int key = 255;
    do
    {
        for (int iDevice = 0; iDevice < num_devices; iDevice++)
        {
            auto pipe = pipelines[iDevice];

            try
            {
                // wait for an available frame
                rs2::frameset frameset = pipe.wait_for_frames();
                sys_time = std::time(nullptr);

                //Get each frame
                //rs2::frame color_frame = frameset.get_color_frame();
                rs2::frame depth_frame = frameset.get_depth_frame();
                rs2::frame ifred_frame = frameset.get_infrared_frame();

                // Creating OpenCV Matrix from a color image
                //cv::Mat color(cv::Size(WIDTH, HEIGHT), CV_8UC3, (void *)color_frame.get_data(), cv::Mat::AUTO_STEP);
                depth16_frames[iDevice] = cv::Mat(cv::Size(WIDTH, HEIGHT), CV_16UC1, (void *)depth_frame.get_data(), cv::Mat::AUTO_STEP);
                depth16_frames[iDevice].convertTo(depth8_frames[iDevice], CV_8U, 1 / 256.0);
                cv::applyColorMap(depth8_frames[iDevice], depth8_frames[iDevice], cv::COLORMAP_JET);
                ifred_frames[iDevice] = cv::Mat(cv::Size(WIDTH, HEIGHT), CV_8UC1, (void *)ifred_frame.get_data(), cv::Mat::AUTO_STEP);

                // Display in a GUI
                std::stringstream ss;
                //cv::imshow("Color", color);
                ss << "InfraRed_" << iDevice;
                cv::imshow(ss.str(), ifred_frames[iDevice]);
                ss.str(std::string());
                ss << "Depth_" << iDevice;
                cv::imshow(ss.str(), depth8_frames[iDevice]);
            } catch (rs2::error e)
            {
                std::cout << "error type: " << e.get_type() << std::endl;
                std::cout << "failed_function: " << e.get_failed_function() << std::endl;
                std::cout << "failed args: " << e.get_failed_args() << std::endl;
                std::cout << "message: " << e.what() << std::endl;
            } catch (...)
            {
                std::cout << "Exception Occured" << std::endl;
            }
        }

        key = cv::waitKey(1);
        std::stringstream ss;
        switch (key)
        {
        case ('s'):
        case ('S'):
            for (int iDevice = 0; iDevice < num_devices; iDevice++)
            {
                ss.str(std::string());
                ss << filePath << sys_time;
                std::cout << ss.str() << std::endl;
                mkdir(ss.str().c_str(), 0755);
                ss << "/" << iDevice;
                std::cout << ss.str() << std::endl;
                mkdir(ss.str().c_str(), 0755);
                //cv::imwrite(ss.str() + "filtertest_color.png", color);
                cv::imwrite(ss.str() + "/filtertest_depth.png", depth8_frames[iDevice]);
                cv::imwrite(ss.str() + "/filtertest_ifred.png", ifred_frames[iDevice]);
            }
            break;
        default:
            break;
        };
    } while (27 != key);

    cv::destroyAllWindows();
}`

sumandeepb, 2019-11-05:
On the same device if I just use:
cfg.enable_stream(RS2_STREAM_ACCEL);

then it works. I get correct IMU sensor data frames. Issue is only when trying to fetch depth / image data stream.

freemanlo, 2019-11-08:
Hi @sumandeepb,

This looks like the USB bandwidth and power capability issue. How about only one D435i connected?
What's kind of platform you used?

sumandeepb, 2019-11-08:
> Hi @sumandeepb,
> 
> This looks like the USB bandwidth and power capability issue. How about only one D435i connected?
> What's kind of platform you used?

hi @freemanlo , there is only one device is connected. I just said that I have 8 D435i in stock, and only 2 of them have this issue. Rest of the 6 work fine. With same version of firmware. Hope that clarifies things.

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

Hi @sumandeepb,

Please check the calibration data (rs-enumerate-devices -c) on these two malfunctioned D435i cameras.
If it's normal, suggest to go back to the reseller for RMA and failure analysis.

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

Hi @sumandeepb,

Did these two D435i cameras' depth streams never work normally before when you got them?

sumandeepb, 2019-11-19:
> Hi @sumandeepb,
> 
> Please check the calibration data (rs-enumerate-devices -c) on these two malfunctioned D435i cameras.
> If it's normal, suggest to go back to the reseller for RMA and failure analysis.

I will check tomorrow and report the results here.

sumandeepb, 2019-11-19:
> Hi @sumandeepb,
> 
> Did these two D435i cameras' depth streams never work normally before when you got them?

I don't think so. We got 8 of them a few months back. Did the development with just 2, they worked fine. Later tested all 8 and out of that 2 never seem to stream the deapth data, only streams the IMU data.

They had really old firmware in them 5.10.xx. Updating the firmware to version 05.11.06.250 (production) did not help. Then again updating to firmware version 05.11.15.00 (development) also did not help.

What could be the reason ?

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi @sumandeepb,

This requires the failure analysis of these two cameras. One reason we faced was that the internal hardware connection between the interposer board and the camera module  is loose due to the accidental drop.
No solid connection will affect the video stream, which requires high data throughput with reliable cable and connection. IMU stream is thru internal I2C interface, which doesn't require the high speed.
By the way, does the stream of these two malfunctioned D435i cameras have the same issue?

RealSenseCustomerSupport, 2019-11-29:
----------------------------------------------

Hi @sumandeepb,

Any update? How about the color stream of these two cameras?
Do you have the RMA?

sumandeepb, 2019-12-03:
> Hi @sumandeepb,
>
> This requires the failure analysis of these two cameras. One reason we faced was that the internal hardware connection between the interposer board and the camera module is loose [...]

I will have to open and check the hardware. I shall update the status in a few days.

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @sumandeepb,

Any update?

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @sumandeepb,

Any update?

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport

AndreV84, 2020-05-28:
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x559ba8ff80):
    Frame didn't arrive within 15000
getting this when running from nvidia jetson or nvidia jetson docker

skribtsov, 2020-08-12:
Same issue with D415, Odroid (Ubuntu 18)