realsenseai/librealsense #5187: How can I get the Depth and Color image registered with the Unity wrapper? Also, is there any Guide I can follow?

mintos002, 2019-11-05:
I've made a function in a .dll that registers 3 images together (color, depth, thermo), for that I need 3 IntPtr.
I need to make a UI useing Unity to show the registered images, so I'm using the RealSense Unity wrapper to pick up the color and depth images from the stream.
I'm following the "AligmentSample" from the "realsense.unitypackage" but I can't find where I can get both images registered (Color and Depth) at the same time.
I've tryed to get them from the "OnNewSample(Frame frame)" function in "RsStreamTextureRender.cs" but the problem is that on the first call of the function I get the color ant in the second de depth, but I need to get both at the same time.

Does anyone know how I can get Depth and Color frames at the same time?
If not, is there any Guide apart from the github readme?

I'll really appreciate any help.
Thanks!

ogoshen, 2019-11-05:
You might be looking for something more along the lines of [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-3-processing/Window.xaml.cs) c# example.

Unity wrapper provides a few abstractions over the .NET wrapper. 
Alignment in the Unity sample is done in the `RsAlign` class [here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsAlign.cs).
The `RsStreamTextureRender` extracts a single video frame, and creates a Unity texture from it.

mintos002, 2019-11-07:
Finally found out that the problem was that I wasn't usint the **using** statments.

Now I have in the UI the depth and the color image streams working perfectly, but now I need to show the RGB thermal image (an `IntPtr`) stream. 
The problem is that it's not a `RsDevice` object so I can't use the `RsStreamTextureRender` or at least that's what I think.
So any idea how I can show the thermal image stream?

Thanks for your help!

closed 2019-11-28 by mintos002