realsenseai/librealsense #5188: finding actual area of an object

edrti, 2019-11-05:
I have pair of rgb and rgbd images, which include an object (a tree leaf).    I was wondering how would I figure out the actual area (size) of the object, using the the captured depth values.

For example, if the realsense camera sees the leaf, and the leaf takes 100 pixels of space in the image (in the rgb image), and the depth value the realsense camera gives is x for each pixel of the leaf in the rgb image, how would I know the actual size of the leaf object?

in addition, if the pair of rgb and rgbd images are aligned, what does it mean? how can i use it?
Thanks in advance.

ev-mp, 2019-11-05:
@edrti, in case you're using D435 there is a ["Edge Map"](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table) preset available that enhances edges and discards surfaces.  Imho it may facilitate extraction of the object's contour, allowing for the area calculation.
[image]

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

@edrti With the suggestion of ev-mp, did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@edrti  If no other questions, will close this. Thanks!

closed 2019-12-13 by RealSenseCustomerSupport

rupak-d, 2021-02-25:
Any chance of getting some more info on how the edge maps are created? Will it see edges when surfaces are similar?