realsenseai/librealsense #5192: Announcement:  RealSense SDK version 2.30.0 now available (many added features and bug fixes)

MartyG-RealSense, 2019-11-05:
Hi everyone,

RealSense SDK version 2.30.0 is now available to download from the link below.  I thought I should give it the status of an Announcement as it is a major release that adds many features and bug-fixes.

https://github.com/IntelRealSense/librealsense/releases/ 

**API CHANGES**

-  RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION - category added to rs2_notification_category. The notification designates T265 performing relocalization event, i.e. finding a match betweenthe current position and a previously recorded map.

**NEW FEATURES & IMPROVEMENTS**

-  New Cross-platform user-space implementation for supported USB protocols named rsusb. The refactored classes replace the multiplicity of UVC device per-platform implementations (Windows, Unix, Android) with a single cross-platform code infrastructure. In order to support the UVC device requirements, some modification were introduced into rsusb API, mainly adding asynchronous API to the USB messenger. Core Features:

-  Asynchronous API added to rsusb (USB request).

*  Multiple implementations of libuvc (Win7 / Linux / Android) replaced with single implementation
HID device modified to work with the new asynchronous API. Impact:

*  A new CMake option named FORCE_RSUSB_BACKEND added replaces the FORCE_LIBUVC and FORCE_WINUSB_UVC flags are now marked as deprecated.

*  An update to WinUSB driver was implemented. The new driver must be applied in order to use librealsense 2.30.0 or newer SDK version on Windows 7. Installation of the new driver is performed via librealsense Windows 7 installer.

- [T265] Firmware Upgrade to 0.2.0.857:

*  Numerical stability improvements in various NaN pose scenarios, can have positive effect on issues #4518, #5101, realsense-ros #955

*  Minor relocalization improvements (including better cross-device map compatibility), toward upcoming greater relocalization update.

-  [T265] New notification category.(RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION), produced on first relocalization to an imported map

-  [T265] display USB port chain in physical port info. by @BriceRenaudeau

-  New face detection and depth-enabled anti-spoofing demo exhibiting machine learning algorithms with DLIB toolkit.

-  [T265] Adding Wheel Odometry python sample.

-  A link to community project that builds Android application with librealsense. @cabelo

-  [Realsense-Viewer] User notification improvement when the rendering format is not supported.

**BUG FIXES**

-  Prevent hex formatting contamination.

-  Acquire depth units from intrinsic. (RS5-5486)

-  On-chip calibration crash fix

-  [rosbag-inspector] Crash fix. (DSO-13665, DSO-13562). Addresses #4704, #4932

-  Memory leak in hid sensor (DSO-13080, DSO-13712, DSO-13639). Fixes #4332

-  Raspbian Buster build fix. Fix IMU streams handling. Addresses #4986, #4979, #4950, #4818.

-  [Realsense-Viewer] Configuration file default path with white spaces was not handled properly (DSO-13701). Fixes #3779

-  Rename foreach to foreach_rs to avoid namespace collisions with QT's "foreach" macro. #4461. Proposed by @cgpadwick's

-  Fix min Z offset for disparity domain colorization mode by @TetsuriSonoda

- [Depth Quality Tool/Viewer] Metrics record fixes and improvements. Fixes #4913, #4948

-  [rs-ar-basic] Fix extrinsic pose to camera transformation

-  Fix White Balance control for Rolling shutter sensor.

-  Robustness improvement: get_distance to verify user-provided pixel indexes. #4877

***********

Full release notes, including all linkages and a list of unresolved Known Issues, are available here:

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2300

mickeyouyou, 2019-11-06:
sounds good 👍

neilyoung, 2019-11-06:
@MartyG-RealSense 
>RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION - category added to rs2_notification_category. The notification designates T265 performing relocalization event, i.e. finding a match betweenthe current position and a previously recorded map.

Wow. Congratulations to that great update. I already tested the relocation feature and it works! I just now need an explanation, what practically can be achieved with this feature. It is just "Hey, I have found, I know this place?", since the translation coordinates are not changing to some value, which has seen in a previous run or what can be done with it?

Besides that there is still this long time error with the file extension (at least on Ubunutu): When you save a map the save dialog suggest to append the .raw extension to the given name. But it doesn't. Trying to reload opens a dialog, which filters for .raw and finds nothing, obviously. You need to change the filter to "all files" in order to see the saved map. Shall I open a separate issue for that?

MartyG-RealSense, 2019-11-06:
Glad you're excited by the update!

I do not have any further information about the workings of RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION at this time.

Yes, please open a separate issue about your file extension problem.  Thanks!

TheBricktop, 2019-11-19:
Will try the update, im really happy that relocalisation feature is coming along, but dissapointed that there is still no support for t265 on android, and it could really benefit our current project as it needs reliable augmented reaity tracking that depends on recorded maps. (we cannot use markers)

dorodnic, 2019-11-19:
Hi @TheBricktop 
#5213 is a significant step in the direction of T265 on Android. Still no commitment we can provide, but a lot of work is being done towards it.

closed 2019-11-19 by dorodnic

GOBish, 2020-01-18:
I still can't get it to work on Raspbian Buster on a RPi4 4GB