realsenseai/librealsense #5193: 2.30.0 and Win7

aparkov, 2019-11-05:
| | |
|---|---|
|**librealsense**|2.30.0 RELEASE|
|**OS**|Windows|

starting with 2.30.0, the viewer or depth Quality tool - Win 7 version - is not recognizing any depth camera (D415, D435).
probably USB3 drivers?

matkatz, 2019-11-06:
Hi, @aparkov 
You are correct, we made a change to the driver that requires driver update.
Please run the [Win7 installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.30.0/Intel.RealSense.SDK-WIN7-2.30.0.1174.exe) with `Intel RealSense Win7 Drivers` flag checked.
We will update the release notes with this issue.

aparkov, 2019-11-06:
the drivers checkbox **was** checked. To be more specific, in the _Device Manager_, the depth cameras are not showing under Imaging Devices; however, under _Universal Serial Bus devices_ there are _D415 Color_ and _D415 Depth_ showing (with the driver dated 12/29/2014, ver 23.56.40.396).

ev-mp, 2019-11-07:
@aparkov hello, are you using the 32 or 64-bit win7 version ?
Can you post the snapshot of device manager view for review ? 
Thank you

aparkov, 2019-11-07:
Win7 Pro, 64-bit.
Device mgr snapshots:
[image]

[image]

dorodnic, 2019-11-10:
Hi @aparkov 
I think manually unistalling device drivers of these two and re-installing the SDK should solve the issue. The fact that it states "D415 Color" instead of "D415 RGB" is an indication that it is still using the old driver.

aparkov, 2019-11-11:
Thank you @dorodnic -- that fixed it!

closed 2019-11-11 by aparkov