realsenseai/librealsense #5194: depth map

gjd2017, 2019-11-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to save depth image with 'White to Black' not 'Jet', how can I do?

MartyG-RealSense, 2019-11-06:
The **rs_measure** sample program demonstrates code in its first lines to set the color scheme with the RS2_OPTION_COLOR_SCHEME instruction.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure#code-overview

gjd2017, 2019-11-06:
@MartyG-RealSense ，Thanks,  solved it!

MartyG-RealSense, 2019-11-06:
Great news - good luck with your project!

gjd2017, 2019-11-06:
@MartyG-RealSense , hi, another questions:  rs-capture（linux-source code）gets different resolution: 480P(clolor-image) and 720P(depth-image)， how can I get the same size of resolution?

MartyG-RealSense, 2019-11-06:
As the 'P' refers to the number of horizontal lines in a resolution, I believe that 720P would be 1280x720, and 480p would be 858x480.

You can set a custom resolution with 'cfg' configuration instructions placed just before the pipe.start() instruction.  Typical instructions for this may be:

// Define the cfg variable

rs2::config cfg;

// Set the resolution and FPS of the depth an color streams

cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 858, 480, RS2_FORMAT_BGR8, 30)

The first two values in the bracket refer to the resolution, and the final number at the end of the bracket (e.g '30') refers to the FPS speed that you want to set the stream to.

gjd2017, 2019-11-07:
@MartyG-RealSense , Thanks your reply, It seems python wrappers, I want C++ wrappers, any way, thank you very much.

MartyG-RealSense, 2019-11-07:
The stream code that I used is C++.  Python versions of stream config instructions tend to use rs.stream instead of rs2.stream, like this Python example:

rs_config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, frame_rate)

I should add that if you use cfg, your pipe start line should be:

**pipe.start(cfg)**

gjd2017, 2019-11-07:
@MartyG-RealSense ， Get it! Thank you very much.

closed 2019-11-07 by gjd2017