realsenseai/librealsense #5195: T265: Unknown SensorType support (0xC)

neilyoung, 2019-11-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-66 generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.30.0 pre-release                   | 
| Language                            |  python                          | 
| Segment			|  others                    | 

### Issue Description
Already noticed this with the current development version, but this seems to be 2.30.0 now, so no wonder. When starting the python `t265_rpy.py` sample with python3 I'm catching two lines like this at the beginning. Then the app seems to work.

```
09:49:10.926 [18236] [E] Device-2C00: Unknown SensorType support (0xC)
09:49:10.926 [18236] [E] Device-2C00: Unknown SensorType support (0xC)

```

bfulkers-i, 2019-12-03:
This one may be fixed now, but will definitely be fixed in #5213

radfordi, 2019-12-06:
I believe this is harmless warning and it will disappear when either #5061 or #5213 are merged.

neilyoung, 2019-12-07:
@radfordi Yes i think I have already said somewhere that i didn’t see this anymore.

closed 2019-12-07 by neilyoung