realsenseai/librealsense #5196: Realsense-Viewer: Minor issue with file name of saved/loaded map

neilyoung, 2019-11-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-66 generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.30.0 pre-release                   | 
| Language                            |  python                          | 
| Segment			|  others                    | 

### Issue Description
When you save a map the save dialog suggests to append the `.raw` extension to the given name. But it doesn't. Trying to reload opens a dialog, which filters for `.raw` and finds nothing, obviously. You need to change the filter to "all files" in order to see the saved map.

dorodnic, 2019-11-11:
Thank you @neilyoung 
We should address it eventually

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

Hi @neilyoung

This thread should now be taken care of by the following PR:

https://github.com/IntelRealSense/librealsense/pull/5370

If so let's go ahead and close the thread.  If it's still open let us know.

neilyoung, 2019-12-25:
Agreed. I will check. Closing this for now

neilyoung, 2019-12-25:
Not fixed, see https://github.com/IntelRealSense/librealsense/pull/5370#issuecomment-568869711

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @neilyoung

Latest RSViewer 2.33.1 saves map with .map extension even if no extension is explicitly applied to filename by user.  When importing map file, the RSViewer shows files with the .map extension for selection by user.

No further updates planned for saving/loading maps.

closed 2020-03-30 by ev-mp