realsenseai/librealsense #5197: Example rs-align-advanced crashes

zeralight, 2019-11-06:
- **Camera Model**: D415
- **Firmware Version**: 05.11.15.00 
- **Operating System & Version**: Ubuntu 18.04
- **Kernel Version**: 5.0.0.32-generic
- **Platform**: PC
- **SDK Version**: 2.29.0
- **Language**: C++

When running the example **rs-align-advanced**, after some time the Desktop crashes and restarts.  
I installed the drivers and the SDK by following the guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md  
I applied the patch **patch-realsense-ubuntu-lts.sh**.  
Also when using realsense-viewer, I see the following log line:  
> WARNING [140545959655168] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

Hi @zeralight,

1. It looks the metadata patch is not applied correctly for Ubuntu 18.04 + kernel 5.0. You can refer to below patch.
https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-metadata-bionic-hwe.patch
2. Do you also see the crash issue when you run other librealsense apps like realsense-viewer, rs-align, etc?
3. Any kernel and librealsense log info when the crash issue happen?

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @zeralight,

Do you try the latest librealsense and still see this issue?

zeralight, 2019-12-05:
Thanks for your assistance.

> 2. Do you also see the crash issue when you run other librealsense apps like realsense-viewer, rs-align, etc?
No, they work perfectly.

> 3. Any kernel and librealsense log info when the crash issue happen?
No, only what I have put on my first comment.

> Do you try the latest librealsense and still see this issue?
No, I installed ubuntu 16 with kernel 4 since then.

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@zeralight,

It's great that your issue resolved based on different Ubuntu/Kernel version.
We would need to take more validation with Ubuntu 18.04/Kernel 5.0.

closed 2019-12-25 by RealSenseCustomerSupport