realsenseai/librealsense #5199: External Power Supply

rofreli, 2019-11-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 5.11.15.0                                                 | 
| Operating System & Version |   Raspbian                                               | 
| Kernel Version (Linux Only)    |  4.19                                                       | 
| Platform                                 | Raspberry Pi                                           |
| SDK Version                           | 2.30.0                                                      | 
| Language                               |  opencv/python                                      | 
| Segment			         |  others                                                    | 

It's possible create some USB cable where can I use an external  power supply from instead my raspi battery. I don't wanna share current and tension and I wanna more independence from each other.

[image]

MartyG-RealSense, 2019-11-06:
A straightforward solution to powering the camera from mains electricity instead of your computing hardware's USB port would be to plug the full-size USB A end of the camera's USB cable into a powered USB 3 hub.  The Intel RealSense team have successfully used the 4-port AmazonBasics powered hub with the 400 Series camera.

https://www.amazon.com/dp/B00DQFGH80/

rofreli, 2019-11-06:
Thank you I will try this solution.

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Is there any question about this ticket? or this ticket would be closed soon.

Thanks

closed 2019-11-19 by RealSenseCustomerSupport