realsenseai/librealsense #5201: Announcement:  Intel blog article on RealSense, mixed reality capture and the MixCast VR system

MartyG-RealSense, 2019-11-06:
Hi everyone,

The RealSense blog has posted an article on Mixed Reality Capture applications (not the same as mixed reality as it is usually understood), and the new MixCast VR system.

https://www.intelrealsense.com/mixed-reality-capture-using-depth-cameras/ 

For further understanding of the concept of RealSense cameras and VR, I recommend researching a 2016 / 2017 prototype mixed reality headset with a RealSense ZR300 stereo camera and in-headset computer that was called Intel Project Alloy.  YouTube has numerous videos on Project Alloy, such as these two Intel demos:

https://youtube.com/watch?v=Be-MJxeCpvU

https://youtube.com/watch?v=Ku9gjx5ECuY

TheBricktop, 2019-11-19:
Such shame that project alloy got slashed, it would be a nice competition to locked in Quest and nonexistent on individual market HTC focus, even better if it would be hackable and open for extensions like maybe with compute cards (another slashed idea). It could work on existent frameworks like WMR or OpenVR or even Monado.

MartyG-RealSense, 2019-11-19:
I believe that Project Alloy was the responsibility of Intel's VR department and not the RealSense department, even though Alloy had a RealSense ZR300 in it.   

Alloy was intended to be a "reference design" - Intel would offer the blueprint for the headset hardware and then manufacturers would produce and sell it themselves instead of Intel producing it.  Alloy coincided with Microsoft's alternative "Windows Mixed Reality" standard, and manufacturers ultimately chose to go with that platform instead of Alloy.

https://www.theverge.com/2017/9/22/16351900/intel-project-alloy-vr-merged-reality-headset-reference-design-discontinued

I don't think it would be too hard to hack together a similar VR headset now, even if it is a bit less cable-tidy than a totally integrated headset unit would be.

TheBricktop, 2019-11-19:
Yeah, but having compute card with i5 and intel UHD gpu would certainly help. This was amazing concept.

MartyG-RealSense, 2019-11-19:
Some cancelled RealSense projects end up coming back in other forms.  2016's cancel of a previewed product was an Android smartphone with a ZR300 called the RealSense Smartphone Developer Kit. Later of course, Android support was successfully introduced for 400 Series cameras.

https://software.intel.com/en-us/blogs/2016/02/22/introducing-intel-realsense-smartphone-developer-kit

closed 2019-11-26 by dorodnic