realsenseai/librealsense #5203: r200 calibration doesn't work

hassan-shehawy, 2019-11-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  R200 | 
| Firmware Version                   | 1.0.72.10 | 
| Operating System & Version |   Win 10  |                                        | 
| Platform                                 | Lenovo ideapad 700  |

### Issue Description
I have very noisy pointlcoud when I run: roslaunch realsense_camera r200_nodelet_rgbd.launch in ROS (my main problem is the flat surfaces appear as non-flat ad for my application, it's critical to distinguish the curves and flat surfaces). So, I followed a recommendation to calibrate the camera using the calibrator app. However, I run it I always get the error: 
An unexpected error has occurred.
Please try restarting your device and run this application again.
Error: DS_INVALID_ARG

MartyG-RealSense, 2019-11-07:
This forum will be the best place to discuss your question, as the Intel Support forum that you also posted at is an old forum no longer responded to by Intel Support staff.   I will re-post my response here.

This error with the calibrator usually occurs on devices with pen support such as Wacom MobileStudio Pro and Acer Aspire Switch 12. An R200 camera typically does not need to be recalibrated.

If your screen has an obstructive window called the 'Intel RealSense Camera Calibration Notifier' when the device starts up, you can stop it appearing by opening the Windows Task Manager interface with the Ctrl-Alt-Del keyboard shortcut, going to the 'Start-up' option tab on the top of the Task Manager window and seeing if the calibrator app is in the list of apps that are set to automatically appear at start-up. If it is in the list, set its status to 'Disabled' to prevent it from launching on start-up.

[image]

hassan-shehawy, 2019-11-07:
Thanks for your response. Disabling it doesn't solve the problem, I still get the same error

MartyG-RealSense, 2019-11-07:
Researching the issue, it seems to have become far more widespread, increasing from a handful of reports a couple of years ago to many on multiple sites across the internet.  It was thought that the problem may be due to the calibrator no longer working with modern versions of Windows 10.  If this is the case, it would help to explain why more and more people have reported this error in 2019 as they discover it for the first time.

MartyG-RealSense, 2019-11-07:
Though calibration can be a factor in image quality, environmental factors such as lighting levels and types of lighting can have an effect on the waviness of a point cloud too.  Could you provide details please about how light it is in the location that the cloud is being generated in (dark / normal / bright lighting) and whether you are using bulb-based lights or ceiling strip-lights.  Thanks!

hassan-shehawy, 2019-11-12:
Thanks for your comment and sorry for late response. I don't know about others, but I don't use Windows unless I have to. Since I started using the R200 I had no problem with the accuracy for all the applications I've been working on. However, for the current project, it's an issue. That's why I was trying to figure out a solution and one of them is of course camera calibration and unfortunately this app is available from Intel only for Windows.

hassan-shehawy, 2019-11-12:
Thanks for your help. Yes, I have been sorting out other variables contributing to the quality and reached the calibration one which saved me before multiple times. For illumination, I tried bulb and LED.
Yes, that's an example of what I get,. The area in the green rectangle should be a flat surface.  

[image]

MartyG-RealSense, 2019-11-12:
Apparently it is possible to do calibration for the R200 using the R200-compatible predecessor of the current RealSense SDK, which is now known as Legacy Librealsense.  This is because, according to one of the Librealsense developer leaders who worked on that Librealsense at the time, rectification is not done on the camera chip, unlike the current 400 Series cameras which do on-board rectification on their Vision Processor D4 chip.

**Calibration discussion**
https://github.com/IntelRealSense/librealsense/issues/463

**Legacy Librealsense**
https://github.com/IntelRealSense/librealsense/tree/v1.12.1

In the past, an R200 user was able to reduce point cloud noise by applying a 'median filter' to the depth image before converting it to a point cloud.

https://github.com/IntelRealSense/realsense-ros/issues/50

hassan-shehawy, 2019-11-14:
Thanks for the clarification.  I think I will try filtering the image and see if it enhances the results.

closed 2019-11-14 by hassan-shehawy