realsenseai/librealsense #5206: Intel Realsense Calibration Algorithm

daothanlong11, 2019-11-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   |  | 
| Operating System & Version |    Linux (Ubuntu 18.04 LTS)  | 
| Kernel Version (Linux Only)    |  5.0                                      | 
| Platform                                 | PC/ NVIDIA Jetson   |
| SDK Version                            |    2.29.0                           | 
| Language                            |  { C++ }                          | 
| Segment			|   Robots                          | 

### Issue Description
Hello Intel Realsense Dev, I want to know the algorithm of how intel realsense camera is calibrated. 
Can you give me some docs or source code about this ? 
Is your algorithm base on the method of Opencv?
Hope to see your answer soon and thanks for all your support

MartyG-RealSense, 2019-11-07:
The 'Programmer Guide' for the RealSense 400 Series' Dynamic Calibrator software tool may be a useful reference for you.

https://www.intel.com/content/www/us/en/support/articles/000026724/emerging-technologies/intel-realsense-technology.html

The guide states: "Users are free to use the example code, but the Intel calibration algorithms contained in the API library is not open source and only available as binary form.  The primary goal is to enable users to develop their own calibration application with the embedded Intel calibration algorithm or their custom calibration algorithms".

Page 13 of the guide, in the section headed 'Dynamic Calibration With Intel Algorithms', may be a good starting point for your research.

daothanlong11, 2019-11-07:
Thanks for your answer so quickly:))
In the question 2, does the camera use any calibration algorithm in the opencv library?

MartyG-RealSense, 2019-11-07:
I hope the link below will be a useful reference for calibration with OpenCV.

https://github.com/IntelRealSense/librealsense/issues/2484

daothanlong11, 2019-11-07:
Thanks for all your answer, this help me a lots :))
I will close this issue

closed 2019-11-07 by daothanlong11