realsenseai/librealsense #5207: T265: lower rate, more robust pose

ajshank, 2019-11-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265 | 
| Firmware Version                   | All | 
| Operating System & Version |   All  | 
| Kernel Version (Linux Only)    |  All                                       | 
| Platform                                 | All  |
| SDK Version                            |  2.29.0+                         | 
| Language                            |  All                          | 
| Segment			|  N/A                      | 

### Enhancement
Does/could the T265 (or its internal firmware/driver) offer support for a more robust pose output, at the expense of lower rate? That is, make poses available at, say 100Hz, but make them even more reliable/precise. I suspect that'll give the onboard algorithms "more time" to achieve better convergence (or run more iterations). Then `librealsense` could perhaps request this rate at start-up.

Maybe, _maybe_, twice the time per loop can help with issues such as https://github.com/IntelRealSense/librealsense/issues/4592 and https://github.com/IntelRealSense/librealsense/issues/4789 (the latter, by perhaps reinitializing features?), and improve accuracy overall?

neilyoung, 2019-11-08:
If it would help to improve things, I would second that. But I doubt that it would. I think the whole engine is clocked by some sensor clock, and since the processing can catch up, this rate is just forwarded. In my experiences, it does also not overwhelm smaller devices (maybe Raspberry Pi Zero excluded) except on the USB interface. Also @ev-mp commented already on the data rate to be expected (https://github.com/IntelRealSense/librealsense/issues/4592#issuecomment-519396899), that should (_should_) basically not be a problem. But I would vote for it. Why having such a high pose rate, if processing systems are fine with less. Of course, one could use just every 5th frame or so, but this is not really an unburden of the communication line...

ajshank, 2019-11-08:
@neilyoung Yes, my interest in this has really nothing to do with overloaded communication interfaces. A lower data rate will probably help with robustness of the poses, and could even potentially cut some power demands (although that might be marginal if at all). Thanks for your vote :)

neilyoung, 2019-11-08:
@ajshank Sure, no doubt. Less power consumption could be a good reason for clocking it down. Would be nice to see if it makes a difference. I have measured a constant power consumption of 400 mA on a banana PI M2 Zero, whereas the power consumption of the camera was only around 150 mA.

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Thanks for the suggestion, at this time the expected behavior for the device is fixed.  I'll close this ticket out but will keep an eye on potential future features if it is possible to do.

closed 2020-01-23 by RealSenseCustomerSupport