realsenseai/librealsense #5209: Depth module in D435 camera stops working

AleBasso80, 2019-11-08:
| Camera Model                       |  Camera D435  | 
| Firmware Version                   | 5.11.15.0 & 5.11.6.250 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra  | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |   2.29.0   | 
| Language                            |  C++ |       
| Segment			|  others   | 

### Issue Description
Hello,
I noticed a very strange behavior in three cameras D435 after about two days of continuous working. The cameras suddenly started and then kept sending valid depth frames, but completely filled with either 0 or the invalid pixel value (I couldn't intercept the issue in debug and my SW sets to 0 the invalid pixel values; I know that the frame is valid from the application log file). At the same time the projector was automatically switched off. The projector temperature communicated by the camera dropped to 0. In the meanwhile the color stream kept on working normally as well as the asic temperature communication.
The other cameras that I have tested, with the same firmware, same SW and same temperature condition, hasn't shown so far after almost one week of continuous working this behavior.

Does someone have already experienced a similar issue or have a clue about what can be happened?

Thanks!

MartyG-RealSense, 2019-11-08:
RealSense 400 Series cameras can run indefinitely so long as they remain within temperature tolerances.  Officially this is 35 degrees C, though it can probably go up to 42 before problems start occurring.  

According to the data sheet document for the 400 Series cameras, "the depth module implements a laser safety control circuit that adjusts laser drive output.  When laser power and depth streaming is enabled and if stereo depth module temperature is >60°, laser power is halved.  If temperature is not lowered below temperature limit within a certain interval, the laser is shut off".

Excessive temperature may also affect the camera's calibration, requiring it to be recalibrated to restore good-quality images.

When problems do occur with long-running, if the camera has remained within temperature tolerances then the source of the problem will tend to be non-camera issues such as a glitch in the computer hardware, operating system or USB port / cable.  It sounds like there was an excessive temperature issue with the cameras in this case though.

AleBasso80, 2019-11-08:
Hello @MartyG-RealSense,
thank you for your quick reply.
What you said was my first thought also considering what the documentation describes, though during last summer I run some very long-running tests (3 months) in worse temperature conditions (environmental temperature about 10°C higher, so that the temperature within the case was higher as well) with the same case and the same acquisition and processing software. The only differences were the cameras, the firmware and the Realsense SDK versions. I didn't experience any issue at all. As a consequence, either different cameras behave differently as regards temperature or the new firmware versions have introduced some changes. Otherwise I don't really know how to interpret this discrepancy.
I can also tell you that after I rerun the cameras, I didn't notice any worsening in the depth image quality and I didn't need to perform a new calibration.

Cheers

MartyG-RealSense, 2019-11-08:
A USB device may experience rapid overheating if there is a fault on the USB port or USB cable.  This kind of problem would typically manifest quickly though, with the casing of the device being hot to the touch within a few minutes.

There was also a past case of a camera temperature rise caused by overheating of a single-board computer, like the Jetson TX2 listed in your specification details at the top of this discussion.

If you anticipate the camera experiencing high temperatures frequently, the new FRAMOS D435e industrial RealSense camera can operate at 55 degrees instead of 35 degrees.  It does cost $1200 USD though.

0 or 'invalid' depth values would suggest that the camera is unable to read any depth detail from the observed scene.

AleBasso80, 2019-11-08:
During the tests I regularly log the Jetson and the realsense temperatures. After a quick ramp I don't notice any other significant change in these values, this is also why I don't know how to relate this issue with temperature, even if it looks very much so. Nevertheless I read on the realsense documentation that the access to the temperature device info should be limited to the development stage. Do you know why? Do you think this might generate some issues in some way?

AleBasso80, 2019-11-08:
PS: I know the new Framos device, unfortunately this is too big for my application.

MartyG-RealSense, 2019-11-08:
I went over all the reported symptoms again.  Another possibility that there may have been some kind of environmental event such as a flash of strong light that saturated the IR sensor, causing it to freeze.  In those circumstances, resetting the camera may correct it.   If you are not physically present all of the time, a **hardware_reset()** instruction can be added to an application's scripting to automatically reset the camera.

I'm not aware of the temperature sensing limitation that you described.

AleBasso80, 2019-11-08:
Thanks, I try to investigate in the direction you pointed out.

As regards the limitation, this is at page 38 of the camera datasheet:
3.6.7 Temperature Sensor
The stereo depth module is equipped with a thermal sensor that is used for laser
safety control (IR Projector). The RealSense library provides access to the thermal
sensor but it is not intended to be used by applications outside of development
environments.

MartyG-RealSense, 2019-11-08:
Thanks for the data sheet reference!  I guess that the information is suggesting that the temperature be treated as debug information.

AleBasso80, 2019-11-08:
Me too. Many thanks for your help. I'll let you know if I figure out what is actually happening.

SirDifferential, 2019-12-09:
Just adding my own findings. We've been observing this issue on many of our D415/D435 sensors over the past year across several firmwares, but I've never been able to reliably reproduce the problem with a minimal example. The symptoms are the same: The depth image suddenly contains largely zero data, although there are a few pixels here and there with valid(?) depth. Restarting the streams fixes this.

Our workaround is to calculate the amount of zero pixels in the depth frame and restart the device when too large percentage of the frame is empty. Some sensors are more likely to suffer from this.

MartyG-RealSense, 2019-12-09:
@SirDifferential  Thanks very much for sharing your method!

AleBasso80, 2019-12-09:
Hi @SirDifferential , 
did you notice whether the projector temperature drops to 0? In our case, when the issue arises we always get this temperature value.

RealSenseCustomerSupport, 2019-12-09:
----------------------------------------------

Hi AleBasso80,

Is there any update regarding the issue?
Like MartyG said, the camera firmware monitors the temperature of laser projector and will shuts it off if it reaches 60°C for several seconds.
Normally, according to lab experiment, if the case (camera skin) temperature is below 44°C, the laser projector temperature should stay below 60°C. Firmware checks the temperature every second while camera is streaming.

IF the firmware reads 0 deg or below, it turns off the laser too. Is there any pattern in your case to reproduce the issue? Or can you try the latest FW from here: https://dev.intelrealsense.com/docs/firmware-releases

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

AleBasso80, 2019-12-10:
Hi,
in my case both the projector and the whole depth module suddenly stop working. A temperature related issue might have sense, however:
- it happens only with some sensors
- it never happened to us with old firmware versions (but perhaps we tested only good sensors, please see point 1)
- if I perform an HW reset, the device restarts working immediately and can work for days without experiencing any new problem. In my opinion this doesn't fit well a temperature related problem.

Thanks

SirDifferential, 2019-12-11:
> did you notice whether the projector temperature drops to 0? I

I haven't been able to get the temperature requests working in my code. The `rs2_sensor` instances report `RS2_OPTION_ASIC_TEMPERATURE` and `RS2_OPTION_PROJECTOR_TEMPERATURE` as supported in `get_supported_options()`, but using `supports()` and `get_option()` fails with:

`failed querying sensor temperature: object doesn't support option #23`

As such I don't know what the temperature readings are.

However, I'm skeptical on if temperature is the problem here: Earlier this year we placed a D435 and an Upboard in a test chamber where we tried various environment temperatures in 20 minute intervals. We went through temperatures from -20°C up to +40°C and only saw anything happening to the realsense in -20°C when the IR emitter turned off. For the rest of the experiment both color and depth streams worked without issues. In the real life scenarios where we've seen the "black depth frame" problem the sensors have generally remained in air conditioned locations where temperatures haven't been nearly as extreme.

> Another possibility that there may have been some kind of environmental event such as a flash of strong light that saturated the IR sensor, causing it to freeze.

Today we tried torturing a D435 with a camera flash and a laser pointer. In both cases realsense viewer clearly displayed the depth feed being influenced by the light as the depth frame went black when the flashes happened. However, the feed quickly returned to normal as soon as the disruption ended. So, we weren't able to reproduce the problem this way. I also would have like to try a burst of sunlight (ie. reflected from an opening glass door), but here in Finland we now have to wait about 4-5 months before we get the next opportunity at that.

At least for us the problem happens pretty rarely, and I don't have a reliable way of reproducing it.

MartyG-RealSense, 2019-12-11:
I wonder if you could simulate a burst of sunlight reflected off an opening glass door by shining a powerful torch onto the door from its outside side (as though light was coming in from outdoors) and then open the door whilst the torch beam is on it.

AleBasso80, 2019-12-11:
Hey all,
I can tell you that we experienced "our" issue even with sensors that weren't exposed to the sunlight or to powerful light source at all.

SirDifferential, 2020-01-03:
I'm curious: Was this the bug that was marked as DSO-11040 in this errata? 

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi AleBasso80,

Is the issue still there after latest firmware update ?

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

AleBasso80, 2020-02-10:
Hello,
I don't have enough data yet to tell. Why do you think that the new fw may
solve the issue? Did you find a bug in the previous one?

Regards
Alessandro

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Hi Alessandro,

Please note, if the issue still exists in latest firmware (or relatively new), the development team will look into it because old firmware issues are not supported after new firmware came out. The issue of DSO-11040 doesn't look like same as what is described in your case, plus it was already fixed in 5.11.2 (Reportedly, you're using 5.11.15.0 & 5.11.6.250).

Did you say 3x cameras had such issue but other cameras didn't (w/ same firmware)? how many cameras do you have? if the failing rate is very high, that must be something else wrong. Do you have a chance to run the 3x cameras on different platform? i.e. Windows 10.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

AleBasso80, 2020-02-18:
Hello,
I can't reproduce the same issue on a different platform, as our application has been developed for Linux and the porting to Windows would require a big effort. I'm aware that I haven't tested many devices, therefore it is difficult to figure out what happened. Nonetheless I have some concerns: before we release any product we test the device for a couple of days and, inevitably, about the 10% of them suffer some instabilities (the issue here mentioned or the need of a HW reset after a few hours of continuous working as all the streams are void). As the tests are run under identical conditions, I started wondering whether somehow the cameras slightly differ from one another, otherwise I have no explanation to this fact. In this thread I mentioned only one specific issue, as it is the most difficult to be dealt with as I am not able to intercept it by means of a grab exception and I didn't want to create a too broad subject. Anyway, my actual concerns are about the whole "picture".
If you have some comments or suggestions, please advise. As regards this particular thread, as far as I'm concerned, it can be closed without a clear outcome. Should I get more information in the future, I would reopen it o create a new one.

Thanks.

MartyG-RealSense, 2020-02-18:
My understanding is that no two cameras have identical extrinsics due to the manufacturing process at the factory but the differences should not be noticeable. The possibility of faulty cameras in a batch also cannot be completely eliminated, though it is extremely rare.  Differences in image quality between multiple units of the same model of camera in the same location are more likely to be related to the calibration of individual cameras.

bellekci, 2020-02-26:
We're experiencing the exact same problem defined here. After a long time, D415 projector is turned off and we do not get any valid frames. Resetting the sensor solves the issue.

We also thought it might be related to the device temperature but it is not. When it occured, the reported temperature was 41 degrees which is the average temperature the device runs reports.

No clue what it is actually related to.

SirDifferential, 2020-02-26:
@bellekci Which firmware version are you using on these cameras? I observed this the other day on 3 sensors, but I forgot to write down their old firmware version before updating to `Signed_Image_UVC_5_12_2_100.bin`. The problem vanished with this new firmware.

bellekci, 2020-02-26:
> 
> 
> @bellekci Which firmware version are you using on these cameras? I observed this the other day on 3 sensors, but I forgot to write down their old firmware version before updating to `Signed_Image_UVC_5_12_2_100.bin`. The problem vanished with this new firmware.

The FW we have is 05.11.15.00. Yours is newer. Do you think we should update?

SirDifferential, 2020-02-26:
Might be worth checking - if you can't reproduce the problem with the new firmware, it would help to isolate this issue further.

AleBasso80, 2020-02-26:
@MartyG-RealSense : the external parameters actually change, in fact I calibrate each single camera, but apart from that different devices seem to have different behaviour.

@SirDifferential and @bellekci : I will try the new firmare too, even unfortunately this is still a development FW.

Thanks!

SirDifferential, 2020-06-05:
I've been experiencing this or similar problem with `Signed_Image_UVC_5_12_3_0.bin` on at least one Raspberry 4 with D435. The depth feed appears under-exposed and does not get fixed until I reboot the application several times. Sometimes it looks like a full system reboot is needed. I don't have a screenshot right now, but I'll supply one when I next experience this issue. The sensor runs most of the day and usually displays this issue in the morning when it has been running for about 15~ minutes and the lights in the office room are turned on.

I just installed `D400_Series_Development_FW_5_12_3_0-1.zip` and will report how it goes.

I did notice the firmware from February 14 in 2019 fixed something similar:

| Number | Status | Errara |
| ---------- | -------- | -------- |
DSO-10428 | Fixed in Development Firmware 5.11.4 | IR Image is black in Auto Exposure (AE) mode with sudden exposure to light

Here are the settings we use with advanced mode:

```
{"aux-param-autoexposure-setpoint":"1536","aux-param-colorcorrection1":"0.298828","aux-param-colorcorrection10":"0","aux-param-colorcorrection11":"0","aux-param-colorcorrection12":"0","aux-param-colorcorrection2":"0.293945","aux-param-colorcorrection3":"0.293945","aux-param-colorcorrection4":"0.114258","aux-param-colorcorrection5":"0","aux-param-colorcorrection6":"0","aux-param-colorcorrection7":"0","aux-param-colorcorrection8":"0","aux-param-colorcorrection9":"0","aux-param-depthclampmax":"65536","aux-param-depthclampmin":"0","aux-param-disparityshift":"0","controls-autoexposure-auto":"True","controls-autoexposure-manual":"8500","controls-color-autoexposure-auto":"True","controls-color-autoexposure-manual":"156","controls-color-backlight-compensation":"0","controls-color-brightness":"0","controls-color-contrast":"50","controls-color-gain":"64","controls-color-gamma":"300","controls-color-hue":"0","controls-color-power-line-frequency":"3","controls-color-saturation":"64","controls-color-sharpness":"50","controls-color-white-balance-auto":"True","controls-color-white-balance-manual":"4600","controls-depth-gain":"16","controls-laserpower":"360","controls-laserstate":"on","ignoreSAD":"0","param-amplitude-factor":"0","param-autoexposure-setpoint":"1536","param-censusenablereg-udiameter":"9","param-censusenablereg-vdiameter":"9","param-censususize":"9","param-censusvsize":"9","param-depthclampmax":"65536","param-depthclampmin":"0","param-depthunits":"1000","param-disableraucolor":"0","param-disablesadcolor":"0","param-disablesadnormalize":"0","param-disablesloleftcolor":"0","param-disableslorightcolor":"1","param-disparitymode":"0","param-disparityshift":"0","param-lambdaad":"751","param-lambdacensus":"6","param-leftrightthreshold":"10","param-maxscorethreshb":"2893","param-medianthreshold":"796","param-minscorethresha":"4","param-neighborthresh":"701","param-raumine":"6","param-rauminn":"3","param-rauminnssum":"7","param-raumins":"2","param-rauminw":"2","param-rauminwesum":"12","param-regioncolorthresholdb":"0.785714","param-regioncolorthresholdg":"0.565558","param-regioncolorthresholdr":"0.985323","param-regionshrinku":"3","param-regionshrinkv":"0","param-robbinsmonrodecrement":"25","param-robbinsmonroincrement":"2","param-rsmdiffthreshold":"1.65625","param-rsmrauslodiffthreshold":"0.71875","param-rsmremovethreshold":"0.809524","param-scanlineedgetaub":"13","param-scanlineedgetaug":"15","param-scanlineedgetaur":"30","param-scanlinep1":"155","param-scanlinep1onediscon":"160","param-scanlinep1twodiscon":"59","param-scanlinep2":"190","param-scanlinep2onediscon":"507","param-scanlinep2twodiscon":"493","param-secondpeakdelta":"575","param-texturecountthresh":"0","param-texturedifferencethresh":"1722","param-usersm":"1","param-zunits":"1000","stream-depth-format":"Z16","stream-fps":"30","stream-height":"720","stream-width":"1280"}
```

MartyG-RealSense, 2020-06-05:
Thanks for the detailed update @SirDifferential - looking forward to your test results!

SirDifferential, 2020-06-15:
Alright, had it happen today after the program had been running for a few hours.

First image: the depth image with the exposure(?) off:
[image]

Second image: after restarting the program, which also causes the streams to be restarted:
[image]

Third image: after rebooting the computer and receiving a valid depth image:
[image]

And for reference, here's the color image:
[image]

MartyG-RealSense, 2020-06-15:
@SirDifferential  Thanks so much for the images.  I considered your situation carefully.  Am I correct in thinking that your third image, described as a 'valid depth image', is what the scene normally looks like when it is working correctly, please?

There have been past reports of the 400 Series camera having problems with grey concrete surfaces that have strong light projecting onto them, producing saturation and therefore an area with missing depth pixels.

https://github.com/IntelRealSense/librealsense/issues/2875

In the first two images with large areas of black, they may give the impression that they are detecting the surface of the objects because the rough shape of the objects is present.  The black represents areas of the image where the camera could not read depth detail though and so they are empty.  An example of this is a depth image of a black cable: the cable may appear on the depth image as a cable but it is really just a cable-shaped area of blank emptiness.

Black and dark grey objects can be hard for a depth camera (not only RealSense) to read detail from because the general principles of physics dictate that the color black absorbs light instead of reflecting it.  Unfortunately you have a lot of black and dark grey objects and surfaces in the scene, as well as the grey concrete floor with a strong light source.

SirDifferential, 2020-06-15:
Yes, the third image is what the image is supposed to look like, and it's what the sensor sees most of the times. We did some exposure testing on my laptop with the same realsense and noticed that the over-exposed image indeed looks like the "problematic" frames:

https://www.youtube.com/watch?v=RKCHzsL7kC4

So this gave us the idea that instead of rebooting the sensor / OS when these problems happen, we'll instead poke the auto-exposure on/off and see if the sensor re-configures its exposure settings properly. The region of interest settings for the exposure might be worth a check as well. 

We'll now wait for this problem to happen again and try out these ideas.

MartyG-RealSense, 2020-07-07:
Hi @AleBasso80 and @SirDifferential   Could you provide updates please about the current status of each of your problems in this case, please?  Thanks!

SirDifferential, 2020-07-07:
I don't have anything to update - the problem hasn't appeared for many weeks and I haven't been able to reproduce it. When it does, I'll spend some time investigating the exposure setting things. This issue can be closed as far as I'm concerned.

MartyG-RealSense, 2020-07-07:
Ok, thanks so much for the quick response, @SirDifferential ... as @AleBasso80 created the case, I will await their response.

AleBasso80, 2020-07-08:
Hello,
I don't have any update and I'm in the same condition as @SirDifferential. As far as I'm concerned the issue can be closed, even if I'm not sure the problem is completely solved.
Thanks.

MartyG-RealSense, 2020-07-08:
@AleBasso80  Thanks so much for the update.  I will close it, but please feel free to open a new issue in the future if you feel it is necessary (not just on this subject but any problem that you have).

closed 2020-07-08 by MartyG-RealSense

dmoham1476, 2021-05-04:
Where can I add the code to reset the camera?  Are there any examples ? I am seeing this for a streaming use-case. The depth frame is becoming 0 causing the picture to be all gray after a stress test of about an hour or two

MartyG-RealSense, 2021-05-04:
Hi @dmoham1476  Here are some links to example code for the **hardware_reset()** instruction.

**C++**
https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

**Python**
https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

dmoham1476, 2021-05-05:
When client code is repeatedly pinging a server where the realsense is hosted, I saw the realsense depth frame becoming zero today again. However it had run only for about ~20 mins. It starts to work again if the machine is fully rebooted and the realsense server code is started again. In this kid of a client<->server streaming use-case setup, would a hardware_reset make sense?

MartyG-RealSense, 2021-05-06:
@dmoham1476   You make a fair point.  The open-source ethernet white paper does provide details about how the device or rs-server can instead be reset using an **ssh** command.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-4-1-common-tasks

dmoham1476, 2021-05-06:
Hi Marty

I also have the realsense + yolov5 object detection running on the same
thread in the flask server. I have commented pipeline.stop() so we dont
have to start, stop the pipeline everytime we call the server.However,
would you suggest otherwise?

Is there any code reference to host only the realsense camera as a FLASK
end point, so I can decouple it from the object recognition model end
point? I believe the hardware reset would affect availability of the server.

Deepa

MartyG-RealSense, 2021-05-06:
Hi @dmoham1476  There is not a large amount of resources relating to using a flask with RealSense, but I have provided a list of links to research below so you can see whether any of them provide information that fits your project needs.  

https://github.com/IntelRealSense/librealsense/issues/6475
https://freesoft.dev/program/155363707
https://github.com/WEMarsSoftware/Cam-Server