realsenseai/librealsense #5210: Save regular to rosbag

Vaja787, 2019-11-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.15.00| 
| Operating System & Version |   Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    | 4.9.140                                         | 
| Platform                                 |NVIDIA Jetson |

### Issue Description
I'd like to modify rs-record in such a way that there is a rosbag created every hour. I'm recording for a few days, so in order to prevend any data losses there should be a bag file created regularly. 

rs-record.ccp does the recording just the way I would like it to have, only the regular saving is not in it. I'm new to programming with ROS and realsense, so can somebody help me modifying this script to let it save a ros bag every hour?

MartyG-RealSense, 2019-11-08:
It sounds to me as though an ideal approach would be to use the ability of ROS to "split" a bag recording, automatically creating a new bag file after a specified duration and continuing on with the recording until the next time that the specified time period is reached.  So after every 3600 seconds of recording (60 minutes), a new bag file could be automatically started.

https://github.com/IntelRealSense/librealsense/issues/2424#issuecomment-423284173

Vaja787, 2019-11-08:
Thanks, that is exactly what I needed, and it works. What I'd like about rs-record is that I can replay the bagfile in the realsense viewer. With ros record --split I'm not able to play back using realsense viewer. Which topics to I need to record to get the same format used by the viewer?

MartyG-RealSense, 2019-11-08:
I'm not sure about this question of topics.  The link below does have a list of topics available to the RealSense ROS wrapper though, if that is any help.

https://github.com/IntelRealSense/realsense-ros#published-topics

Vaja787, 2019-11-08:
@MartyG-RealSense I learned that the rosbag produced by the viewer is not really a real rosbag when you compare it to the rs-record. Is there a way to integrate the --split to rs-record? and if so, where should it be added?

MartyG-RealSense, 2019-11-08:
This subject is outside of my ROS knowledge, unfortunately.  One of the Intel RealSense team members who support this forum are likely to be able to offer good advice for your question.  I appreciate your patience.

Vaja787, 2019-11-08:
@RealSenseCustomerSupport Can you help me out to add the split function to rs-record?

Vaja787, 2019-11-10:
@MartyG-RealSense Maybe you can help me find a solution. My standards dropped over the weekend. Here is why: I tried recording with the viewer for a day, just to see if it was able to record for so long. It worked, however, this gets me stuck with a .bag file of 1.2 TB.... Unable to play it anywhere.... 

So, is there any method that lets me record smaller bag files? The --split function is not working for me. The buffer exceeds really rapit. Even making small chunksizes, small durations and increasing buffer doesn't work. When I record for 3 seconds, the buffer is already exceeding with an infinte buffer...

MartyG-RealSense, 2019-11-10:
One way to reduce bag file size when using rs-record-playback is to enable compression.  The lengthy discussion in the link below looks at this subject.   Also, in the RealSense Viewer, you can enable compression in the playback and record preferences in the Viewer's settings window.

https://github.com/IntelRealSense/librealsense/issues/4499

Vaja787, 2019-11-10:
@MartyG-RealSense Thank you for your amazingly quick response! If I look at this, I can't really relate. My goal is to record for 3 days, resulting in 3 TB of data. If compression even accomplish to have half the file size, I still have 1.5 TB. 

I found issue [#4633](https://github.com/IntelRealSense/librealsense/issues/4633) very interesting. And I think this might provide me with a solution. Where the issue gives information to record for a few seconds and waits for movement again. I just need to record for 15min/30min/1hour (whatever suits the script) and start recording immediately after. Unfortunatly, this is beyond my knowledge on how to implement...

MartyG-RealSense, 2019-11-10:
You are very welcome.   :)

The link below discusses a case similar to what you have in mind.  The user only wants to record when motion is detected across the camera centre-line in order to avoid recording a huge bag.   Dorodnic the RealSense SDK Manager gives advice on how this might be implemented.

https://github.com/IntelRealSense/librealsense/issues/2433#issuecomment-423957626

I am out of office now for an hour but will happily reply to any comment you leave below as soon as I return.   Good luck!

Vaja787, 2019-11-10:
@MartyG-RealSense  I tried and I tried, without success :( I'm looking into the script of rs-record. I'm unfamiliar with C++ so it is difficult for me. However, I'm thinking, the scripts starts recording, and stops it after a given amount of time. Can it made into a loop, that it starts and stops recording in a loop, untill I say to it to stop it? I think this method makes most sense? But yet again, I can't program at this level in C++, so can anybody help me with this? 

PS: how to make changes to rs-record? I edit the version located at librealsense/tools/recorder. If I change something simple as the time to record or the name of the bag file, the changes don't take affect? So I think I'm editing the wrong file.

MartyG-RealSense, 2019-11-10:
I apologize for the delay in responding further.  I have been trying to think of.a way to solve this problem without C++ programming and as easily as possible.  My recommendation is to forget about rs-record-playback and instead record with the ROS bag split.  You should be able to play the bag back with ROS's own viewer tool called **rqt_bag**.

http://wiki.ros.org/rqt_bag

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

@Vaja787 Did you get it through? Looking forward to your update. Thanks!

closed 2019-12-10 by dorodnic