realsenseai/librealsense #5212: D435i Motion Module Hardware Timestamps using LIBUVC

JeremyBYU, 2019-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  Linux pop-os 5.3.0-20-generic                                | 
| Platform                                 | PC |
| SDK Version                            |  2.30                      | 
| Language                            |  C++                       | 
| Segment			|  Robots                     | 

I have installed the latest Realsense SDK 2.30 on my machine. Here is CMAKE commands for completeness: 
```
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true
```

My kernel just upgraded to 5.3 so I now need to use the libuvc backend. It installs and seems to work fine. I have already calibrated the IMU using intel procedures. Hardware timestamps seem to be working for RGB and depth. However I am not **sure** if they are working for the motion module. There are **no errors** about meta information in the terminal when running realsense-viewer.  There is this error but not sure if its relevant, they are few of them over the lifespan, and mostly in the beginning.

```
08/11 10:51:57,716 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:51:57,776 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:51:57,957 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:51:58,138 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:51:58,319 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:51:58,379 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 10:52:16,873 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 11:04:04,052 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/11 11:08:09,032 WARNING [139733166438144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
However when I click the meta information overlay for the motion I get the following picture.

[image]

As you can see the clock domain is showing Global Time, which should mean that hardware timestamps are working. However the backend timestamp is showing 0, making me think that its not working. The main question I have:

1. How do I know if the hardware timestamps are working?
2. And *if* they are not working what must be done to get libuvc to work with them?

A final question I have is a little general. I have searched for it but have not foudn the answer:

1. What are the advantages and disadvantages of using libuvc vs patched kernel modules?

Obviously ease of installation is for libuvc. But are there any tradeoffs? Like higher latency, more bugs, unsupported workflows? Any info on that would be great.

Thanks again for the great product and the support Intel provides.

argosvr, 2019-11-08:
same problem here...

ev-mp, 2019-11-10:
@JeremyBYU hello, please see answers inlined:
>My kernel just upgraded to 5.3 so I now need to use the libuvc backend

With all kernels v.416+ you get the standard HW timestamp for free (w/o kernel patches), You'd still need to apply the patches if you need the extended frame metadata - exposure, gain, laser poser, etc'. But if you're only interested in getting HW timestamps then using unmodified kernel should suffice.

>As you can see the clock domain is showing Global Time, which should mean that hardware timestamps are working. However the backend timestamp is showing 0, making me think that its not working.

The `backend_timestamp` is host-generated attribute. Global timestamp does relies on HW timestamp and is not affected by backend ts. There is a minor issue with not all metadata attributes  being display in the viewer that we'll need to check.

>How do I know if the hardware timestamps are working?

By checking the `timestamp domain` attribute of frame's timestamp. When HW timestamps are available then the SDK will automatically prefer them over the host-generated timestamps and you're expected to receive `global ts` or `hw ts` domain. ([documentation](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a21c0e6615083c0a8d667a05adf6db03e)). Also see related #4525.

>And if they are not working what must be done to get libuvc to work with them?

The HW timestamps should be available with `rsusb` (`libuvc` is deprecated with v2.30.0) when that backend is configured in Cmake. No additional parameters are required.

>What are the advantages and disadvantages of using libuvc vs patched kernel modules?

This questing is worth a topic of its own:), The following points in my estimation are valid to the discussion:
Libuvc advantages:
- Cross-platform  (at least for Linux & MacOS).
- User-space UVC implementation :
     - No kernel patches required
     - Easily to deploy (gcc + libusb dependencies only) - thus is the preferred choice of Librealsense for ARM/Jetsons/Mac platforms
     - Easier to Debug
Disadvantages/ Fundamental issues:
 - **No official support/maintainers.**
 - Most of the implementation is build around UVC 1.1, while most contemporary cameras (including realsense) are UVC 1.5 devices 
-  User-space UVC implementation:
      - Power-management - In case an application crushes/get stuck the the underlying device is not being released and continues to run in `orphan` mode. This may require manual re-plugging to recover (reliability).
- Single Consumer - most kernel drivers (Linux/Windows) allow to connect and communicate with device from multiple processes (except for streaming). With Libuvc only one application can get device handle. This is one of the limitations for multicam on MacOS, for instance. There are [attempts to address this](https://github.com/gregkh/usbutils/pull/52) by community enthusiasts, so this may change one day.

Kernel patches:
- The kernels are fast and (mostly) stable. Adding patched modules into kernel tree in controlled and pier-reviewed manner allows to get most of the benefits and also get the additional features.
- All kernel patches should eventually get upstreamed into Linux of abandoned,
From the user's perspective the main obstacle in accepting the kernel patch model is the deployment scheme that requires certain engineering skills level.
We address this by by wrapping and redistributing the patches with DKMS Debian package, at least for Ubuntu LTS kernels.

JeremyBYU, 2019-11-11:
Thank you for this thorough response. I will continue to use the libuvc/rsusb driver for now, since mostly all I care about is timestamps.  The only thing that you mentioned that I am a a little concerned with is this quote:

> The kernels are **fast** and (mostly) stable.

Just wondering if this implies that there might be an issue with speed when using libuvc.  However that is something that I can determine myself through my own tests.

JeremyBYU, 2019-11-11:
I think I just ran into an issue with something you just said:

> Single Consumer - most kernel drivers (Linux/Windows) allow to connect and communicate with device from multiple processes (except for streaming). With Libuvc only one application can get device [...]

I have a program that is modified from rs-motion.  Two rs:pipelines are created

1. One pipeline that monitors motion data through a callback. e.g. pipeline.start(config, callback). This puts it in a seperate thread.
2. Another pipeline that gets aligned rgb, depth camera data.  This is on the main thread using the blocking wait for frames: rs2::frameset data = pipe_camera.wait_for_frames();

This setup works just fine on my Windows 10 computer (which I assume is using a good device driver that supports this setup). However on my Linux computer now using libuvc/rsusb, only the last pipeline started gets access to data (camera data). Code attached at bottom.

If you comment out the `auto profile_camera = pipe_camera.start(cfg_camera);` line and any references to it a few lines below, then motion data is now recorded.

I just want to make sure that this is not a bug, and this "issue" is what you were referring to in the above quote. If it is expected behaviour, do you have any solution for me to capture high frequency motion data on on thread, and separately capture a frame set of depth/rgb on another using rsusb? I think I have an idea on how to do that and will try it out: Capture all data in one pipeline, each callback gives back each sensor stream independently (gryo, accel, depht, rgb). But then I think I lose the ability to nicely use frameset alignment and such.

Thanks for your help @ev-mp.

```cpp
// License: Apache 2.0. See LICENSE file in root directory.

// Copyright(c) 2019 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp>

#include <mutex>

#include "example.hpp" // Include short list of convenience functions for rendering

#include <cstring>

#include <chrono>
#include <thread>

struct short3

{

    uint16_t x, y, z;
};

#include "d435.h"

void draw_axes()

{

    glLineWidth(2);

    glBegin(GL_LINES);

    // Draw x, y, z axes

    glColor3f(1, 0, 0);
    glVertex3f(0, 0, 0);
    glVertex3f(-1, 0, 0);

    glColor3f(0, 1, 0);
    glVertex3f(0, 0, 0);
    glVertex3f(0, -1, 0);

    glColor3f(0, 0, 1);
    glVertex3f(0, 0, 0);
    glVertex3f(0, 0, 1);

    glEnd();

    glLineWidth(1);
}

void draw_floor()
[... 515 lines cut ...]

    pipe_motion.stop();

    return EXIT_SUCCESS;
}

catch (const rs2::error &e)

{

    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;

    return EXIT_FAILURE;
}

catch (const std::exception &e)

{

    std::cerr << e.what() << std::endl;

    return EXIT_FAILURE;
}

```

JeremyBYU, 2019-11-11:
To make a clarification on why I am actually still using rsusb. My final deployment will be on a Raspberry Pi4. This seems to the recommended way to install realsense. I have been switching between Linux x86_64 (personal PC) and RPI4 buster x64 (experimental new 64 bit kernel)  and testing code between the two. It seems to be working okay except with this new issue.

JeremyBYU, 2019-11-12:
All my investigations have determined that it will not be possible to use Libuvc/rsusb to get high frequency motion data (200fps) at the same time of getting low frequency video data (30fps).  I must use kernel drivers.

It was previously said:

> With all kernels v.416+ you get the standard HW timestamp for free (w/o kernel patches), You'd still need to apply the patches if you need the extended frame metadata - exposure, gain, laser poser, [...]

My investigations are showing this not to be true.  I will open up a separate issue for that. Closing this now.

closed 2019-11-12 by JeremyBYU

ev-mp, 2019-11-27:
The [following comment](https://github.com/IntelRealSense/librealsense/issues/5315#issuecomment-559054292)  is imho relevant to the discussion (#5315)

JeremyBYU, 2019-11-27:
I believe this comment indicates that the libuvc/rsusb driver did not work for the customer. They highlight several issues and say they used the v4l2 (kernel) driver instead and the issues went away. this is what I am trying to do as well now. Unfortunately I have run into these issues with the kennel driver: #4505

JeremyBYU, 2019-12-03:
I have found away to get high frequency motion data at the same time as low frequency video using the RSUSB (LIBUVC) drivers.  Below is code that demonstates this technique while reading bag file as well as live streaming.  The main technique that makes this possible is to NOT USE rs::pipeline, but to directly acess each sensor and configure the stream manually (as well as associated callbacks).

```cpp
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
// After many many times of trial and error I have finally arrived on this technique to get multiple sensor streams from
// an intel realsense which have DIFFERENT FRAME RATES

// This technique works with RSUSB driver (LIBUVC) but should work with kernel drivers as well
// This example is meant for D435i, to get high frequency motion data at the same time of depth,rgb images
// Two examples are shown, one that read a bags file and another that does live streaming from the sensor. Activate bag reading with cmd line flag --bag=<file>
// The main technique that makes this possible is to NOT USE rs::pipeline, but to directly acess each sensor and 
// configure the stream manually (as well as associated callbacks).

#include <cstring>
#include <chrono>
#include <thread>
#include <mutex>
#include <map>
#include <string>
#include <sstream>
#include <iostream>
#include <iomanip>

#include <librealsense2/rs.hpp>
#include <gflags/gflags.h>

using namespace std::chrono_literals;

// Command line flags
DEFINE_string(bag, "",
				"Path to bag file");

bool check_imu_is_supported()
{

	bool found_gyro = false;
	bool found_accel = false;
	rs2::context ctx;
	for (auto dev : ctx.query_devices())
	{
		// The same device should support gyro and accel
		found_gyro = false;
		found_accel = false;
		for (auto sensor : dev.query_sensors())
		{
			for (auto profile : sensor.get_stream_profiles())
			{
				if (profile.stream_type() == RS2_STREAM_GYRO)
					found_gyro = true;
				if (profile.stream_type() == RS2_STREAM_ACCEL)
					found_accel = true;
			}
		}
		if (found_gyro && found_accel)
			break;
	}
	return found_gyro && found_accel;
}

int bag_counter(std::string file_name)
{

	size_t gryro_iter = 1;
	size_t accel_iter = 0;
	size_t depth_iter = 0;
	size_t color_iter = 0;
	double ts_gyro = 0.0;
	double ts_accel = 0.0;
	double ts_depth = 0.0;
[... 234 lines cut ...]
	{
		bag_counter(FLAGS_bag);
	}
	
	
}
catch (const rs2::error & e)

{

	std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;

	return EXIT_FAILURE;
}

catch (const std::exception & e)

{

	std::cerr << e.what() << std::endl;

	return EXIT_FAILURE;
}

```

sunlong169, 2022-06-25:
> You'd still need to apply the patches if you need the extended frame metadata - exposure, gain, laser poser, etc

Which patch is about `exposure, gain, laser poser`?Could you please tell me the name? I carefully checked the files whose postfix is `.patch`, but I did not find it in the `librealsense`.