realsenseai/librealsense #5214: T265: Still a few more doubts on apriltag pose detection sample....

neilyoung, 2019-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-66 generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.30.0 pre-release                   | 
| Language                            |  C++                          | 
| Segment			|  others                    | 

### Issue Description
Oh, man, I have so many issues open, please forgive me if it sucks. But I'm gaining more and more land in this interesting matter, and a lot of questions inevitably arise.

This is again one of the somewhat longer to formulate kind, sorry for that. There is also no `tl;dr`, unfortunately. 

Essentially, it is a request for confirmation/negation of some observations regarding the recognition and positioning of Apriltags with the aim of determining their position in a training, storing it (set_static_node) for later in the map and using the annotated map then for tracking (get_static_node).  The Apriltags are supposed to carry additional context information. This should be sufficient for the formulation of the rough target. I don't know, if I will reach it, but I try :)

So I took the `rs-pose-apriltag` example and the `rs-ar-basic` example and combined both, quick and dirty. Basically I won a quaternion and a translation vector from the two poses in the `rs-pose-apriltag` example, namely the `tags.pose_in_camera[t]` and the `tags.pose_in_world[t]`. Then I coded these values into into the function `reset_object_pose` in the `rs-ar-basic` example in order to make it visible.

The AprilTag was glued to the door, both scripts were running from the same position one after the other.

Here is the result, shown is the "world pose".  The first shot was taken with the laptop on the ground, roughly a meter away from the tag. All others have been taken from the corner of my desk, roughly 1.5 m away and 1 m high. And a little left shifted. And door opened :)

For the first shot, the pose is pretty ok. It looks as like the middle line of the **lower border** of the rectangle is the point (the next pictures will show something different, as you will see). At least I could walk around and the pose was stable. It was not the exact distance, but it was acceptably.

[image]

These are the values, which formed that image:

```
frame 726|tag id: 0|camera R: 0.998, 0.002,-0.069,-0.004, 1.000,-0.024, 0.069, 0.024, 0.997,|t:-0.036,-0.315, 0.914,
float f[7] = {-0.0358145,-0.315414,0.913617,0.0118154,-0.0343299,-0.00165176,0.999339};
                    world  R: 0.996,-0.021,-0.083,-0.023,-1.000,-0.022,-0.082, 0.023,-0.996,|t:-0.078, 0.273,-0.929,
float f[7] = {-0.0778453,0.272905,-0.929469,0.999022,-0.011028,-0.0413003,0.0112858};
```
An explanation of the values: Besides the three inbuilt traces (frame, camera and world), which show the frame number and the rotation matrix and translation vector of both poses, there are two "float" arrays, which I generate and print from the altered `rs-pose-apriltag` example in order to put it into the `rs-ar-basic` example.

The array contains the translation x, y and z values first, followed by the quaternion x, y, z and w derived from the rotation matrix. I'm pretty sure, the "Rotation matrix to quaternion" conversion, I'm using, is correct. I have double checked it. The first array describes the camera pose, the second the world pose.

The second run (from the desktop corner) showed a different picture: Now the point was the middle of the **upper border** of the rectangle and the z-axis (blue) was shown differently. Hmm..  Also the distance error was a bit higher, but still acceptable. I would not make the distance a problem. But the different orientation of the z-axis and the different recognition base could be one, even though maybe just a small one.

[image]

Values now:

```
frame 1092|tag id: 0|camera R: 0.920,-0.001, 0.392,-0.029, 0.997, 0.071,-0.391,-0.076, 0.917,|t: 0.249, 0.444, 1.062,
float f[7] = {0.248763,0.443619,1.06177,-0.0375821,0.19976,-0.00727384,0.979097};
                     world  R: 0.925,-0.043, 0.378,-0.040,-0.999,-0.016, 0.378,-0.001,-0.926,|t: 0.180,-0.368,-1.098,
float f[7] = {0.180378,-0.368166,-1.09798,0.980998,-0.021388,0.192796,0.00386514};

```

**Anything obvious, which would explain the different center and the axis orientation?**

Then I made two closer shots, which show the delta in distance:

[image]

[image]

The camera pose, btw, visualized in both cases to something pretty unexpected, but after some thoughts, I think it is OK. It is not the actual camera pose, but the tag's pose in the camera's coordinate system.

[image]

[image]

It is noticeable that the axis orientation is correct here.

So my questions: 
- Is all that plausible at all? 
- Why is the detection choosing different edges of the tag in different runs?
- Why is the axis orientation of z different and can this be a problem?

I will deal with this the next days and report, if I get answers by myself.

Thanks

ajshank, 2019-11-10:
@neilyoung If I recall, for apriltags, the _center_ of the tag should be the center of the frame (and not one of the edges). Have you had good results with this tag family in general (with, say, the base apriltags library)? I've seen that some tag families have weird detection issues/offsets.

As for your orientation of z-axis, it looks to me that z (blue) is always pointed into the tag in all your images. In fact, the quaternions you reported in your float array for open & closed door scenarios are actually pretty close (and might even correspond to the angle of your door!) :)

neilyoung, 2019-11-11:
@ajshank Thanks for commenting

>If I recall, for apriltags, the center of the tag should be the center of the frame (and not one of the edges). 

I would expect this too, somehow. But meanwhile I have seen a couple of other detection frames. Most are centered at one of the upper or lower borders, but I also did see one, which was nailed to the bottom/left corner. But none of it have been positioned in the center of the tag so far.

>Have you had good results with this tag family in general (with, say, the base apriltags library)? I've seen that some tag families have weird detection issues/offsets.

I have absolutely no experiences with ATs. I just took the one which was bundled with the sample. Will try another family.

> As for your orientation of z-axis, it looks to me that z (blue) is always pointed into the tag in all your images.

Wow, you have pretty good eyes. I have totally overseen that in the analysis I did later on. In fact I found yet another screenshot of the first pose setup (with the closed door), and it clearly shows, that the z-axis **on all pictures** points "into" the tag. It was a perspective issue, obviously. Here the image I found later:

[image]

> In fact, the quaternions you reported in your float array for open & closed door scenarios are actually pretty close (and might even correspond to the angle of your door!) :)

That I true. The angle of the door was always nicely reflected. 

Basically, this angle is finally, what I'm interested in. I'm experimenting with the yaw value (rotation around the green y axis). But I'm not sure if this will help me. Do you know of any magic math, which would allow me to determine the angle between the camera and the tag's z axis.

To be a bit more specific: I would like to know the angle marked with a question mark in the image below. I would estimate it to be around 180 degrees, and around 170 degrees in the photo above the drawing, but I'm missing the correct math here. 

[image]

EDIT: I think my drawing is not entirely true. The ray angle at the camera must not necessarily be 90 degrees... But I also have the yaw value of the cam in the coord system of the cam, if that would help...

ajshank, 2019-11-11:
I've forgotten most of the AprilTags-related things I knew, and I don't have a setup lying around that I can quickly test with, sorry. I'm also not very familiar with all the extra stuff that librealsense can report for AprilTags.
But here are a two things to narrow down your problem:

1. The angle that you marked with a question mark, should be in 2D, i.e., there should be two angles: an elevation and an azimuth. The way you drew it, you are considering a rotation only about y-axis, which would of course make the angle bigger or smaller. There is another possible rotation about x-axis. Perhaps it is easier if you assume in the beginning that xz-plane of the tag is level with the camera, so that you can start with 1D.
2. The 3D location of the tag (relative to the camera frame) is enough to calculate the angle you've marked with a dot, since you can consider the camera to be at "origin" and use tag's `x, y, z` to compute this angle. Then the roll, pitch and yaw angles of the tag (in the camera frame) can be added/subtracted to this angle to get what you need.

neilyoung, 2019-11-11:
@ajshank Thanks again for your input. I totally agree with 1) I also saw this little problem arising. I would just change 2D to 3D and 1D to 2D in your comment.  I also agree on 2) I have all the information there. I should be able to crunch that.

In the meantime I have really merged that two applications into one. If I would have a bit more C++ knowledge, I would consider to contribute that to the community. Basically I have now an app, wich does no longer need my manual copy&paste of the found world coordinates of the tag, but one, which displays the found position "as it comes". And you know what I'm seeing? The anchor point is exactly the center of the tag, as you commented. It was just a matter of the sample coordinate I have chosen from the many, which gave me the wrong impression.

As far as I know there is no inbuilt support for ApritTag detection in librealsense.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @neilyoung

Is there anything further necessary on this thread?

neilyoung, 2020-01-27:
No. Can be closed.

closed 2020-01-27 by neilyoung