realsenseai/librealsense #5215: How to plot point cloud data using matplotlib's scatter plot (3D)

Saminhasan5220, 2019-11-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | {05.11.15.00}| 
| Operating System & Version |   {Win (10) }| 
| Platform                                 | {PC}|
| Language                               |  {python} | 
| Segment			        |  {Robot} | 

### Issue Description
HI,
I have rc car with raspberry pi which i want to drive autonomusly in my living room avoiding static obstacles. I get position of the car using aruco makers  taped to wall at fixed opisition. In order to avoid static obstacles, i want ot make a heightmap from the d435 and use it with a star to genarate a path.

So to summerize, My question is , how to plot 3D point cloud(without texture) in matplotlib from the d435  where the camera is the reference frame.

Sorry, if my question is stupid or something ,i don't have much experience in programming.
Follwing is a sample code of how i intend to use it.

`import cv2                        
import numpy as np                      
import pyrealsense2 as rs
import matplotlib as mpl
import matplotlib.pyplot as plt          
import time
from mpl_toolkits.mplot3d import Axes3D
fig1 = plt.figure(1)
ar = fig1.gca(projection='3d')
ar.set_xlim(-1, 1)
ar.set_ylim(-1, 1)
ar.set_zlim(-1, 1)

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipe_profile =pipeline.start(config)

points =[[[]]]
def get_3dPoints():*****#Function to be written
    pass

try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        if not depth_frame or not color_frame: #or not infrared_frame:
            continue
        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        colorizer = rs.colorizer()
        depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
        cv2.imshow('RealSense', depth_colormap)
        """
        points = get_3dPoints()
        tgt = ar.scatter(points, 'red')
        ar.set_xlabel('X - axis')
        ar.set_ylabel('Y - axis')
        ar.set_zlabel('Z - axis')
        plt.pause(0.00000000000000001)
        tgt.remove()
        """
        cv2.waitKey(1)

finally:

    # Stop streaming#img = cv2.resize(img, (1280, 720), interpolation=cv2.INTER_AREA)
    pipeline.stop()
`
`

dorodnic, 2019-11-11:
Hi @Saminhasan5220 
It's a valid question. I'm not an expert in python to help with this, but I think you are on the right track and the two pointcloud examples may help - [opencv_pointcloud_viewer.py](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/opencv_pointcloud_viewer.py) and [pyglet_pointcloud_viewer.py](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/pyglet_pointcloud_viewer.py)

Saminhasan5220, 2019-12-02:
Solved

thanks to @dorodnic 

`
   from mpl_toolkits.mplot3d.axes3d import *

    import matplotlib.pyplot as plt

    from scipy.interpolate import griddata

    import pyrealsense2 as rs

    import numpy as np

    fig1 = plt.figure(1)

    ar = Axes3D(fig1)#fig1.gca()#

    ar.set_xlim(-2, 2)

    ar.set_ylim(-2, 2)

    ar.set_zlim(-2, 2)

    index = 0

    pipeline = rs.pipeline()

    config = rs.config()

    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    pipeline.start(config)

    color = False

    profile = pipeline.get_active_profile()

    depth_profile = rs.video_stream_profile(profile.get_stream(rs.stream.depth))

    depth_intrinsics = depth_profile.get_intrinsics()

    w, h = depth_intrinsics.width, depth_intrinsics.height

    pc = rs.pointcloud()

    decimate = rs.decimation_filter()

    decimate.set_option(rs.option.filter_magnitude, 3)

    colorizer = rs.colorizer()

    x = []

    y = []

    z = []

    while True: 

        frames = pipeline.wait_for_frames()

        depth_frame = frames.get_depth_frame()

        color_frame = frames.get_color_frame()

        depth_frame = decimate.process(depth_frame)

        depth_intrinsics = rs.video_stream_profile(depth_frame.profile).get_intrinsics()

        w, h = depth_intrinsics.width, depth_intrinsics.height

        depth_image = np.asanyarray(depth_frame.get_data())

        color_image = np.asanyarray(color_frame.get_data())

        depth_colormap = np.asanyarray(

        colorizer.colorize(depth_frame).get_data())

        if color:

            mapped_frame, color_source = color_frame, color_image

        else:

            mapped_frame, color_source = depth_frame, depth_colormap

        points = pc.calculate(depth_frame)

        v, t = points.get_vertices(), points.get_texture_coordinates()

        verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz

        ptss = np.asanyarray(v).view(np.float32).reshape(len(verts),3)

        for i in range(0,len(ptss),5):

            if(ptss[i][2]< 1.5 and ptss[i][1]> -1 and  ptss[i][1]< 2):

                x.append(ptss[i][0])

                y.append(ptss[i][1])

                z.append(ptss[i][2])

        xj = np.asarray(x)

        yj = np.asarray(z)

        zj = np.asarray(y)

        xi = np.linspace(min(xj), max(xj))

        yi = np.linspace(min(yj), max(yj))

        X, Y = np.meshgrid(xi, yi)

        print(len(xj))

        Z = griddata((xj, yj), zj, (X, Y), method='nearest')

        ar.contour(X, Y, Z)

        surf = ar.scatter(xj, yj, zj,s=1,c='r')

        ar.set_xlabel('Horizontal - axis')

        ar.set_ylabel('Depth  - axis')

        ar.set_zlabel('Vertical - axis')

        plt.pause(0.00000000000000000000000000000000000000000000000001)

        surf.remove()

        ar.clear()

        x = []

        y = []

        z = []`

closed 2019-12-02 by Saminhasan5220