realsenseai/librealsense #5216: error: no matching function for call to 'rs2::frame::as()

Andrew-fedoralabs, 2019-11-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857| 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.18                                        | 
| Platform                                 | PC |
| SDK Version                            |  2                        | 
| Language                            |  C++                         | 
| Segment			|  Robotics                        | 

### Issue Description
I get the following issue while following the rs-pose example https://dev.intelrealsense.com/docs/rs-pose

error: no matching function for call to 'rs2::frame::as()'
   auto pose_data = f.as().get_pose_data();
                         ^
In file included from /usr/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /usr/include/librealsense2/hpp/rs_device.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/include/librealsense2/rs.hpp:9,
/usr/include/librealsense2/hpp/rs_frame.hpp:573:11: note: candidate: template<class T> T rs2::frame::as() const
         T as() const
           ^~
/usr/include/librealsense2/hpp/rs_frame.hpp:573:11: note:   template argument deduction/substitution failed:
couldn't deduce template parameter 'T'
   auto pose_data = f.as().get_pose_data();
                         ^

ev-mp, 2019-11-10:
@Andrew-fedoralabs , there seem to be inconsistency between 
https://dev.intelrealsense.com/docs/rs-pose and
https://github.com/IntelRealSense/librealsense/tree/master/examples/pose with the latter being the source.
Thanks for reporting - it will be rectified. 
Till resolved follow the description in the [origin](https://github.com/IntelRealSense/librealsense/tree/master/examples/pose#code-overview)
```c++
// Cast the frame to pose_frame and get its data
auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
```

dorodnic, 2019-11-11:
It seems the problem is with how ReadMe.io is handling GitHub code snippets. I have updated the relevant document to render correctly

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

@Andrew-fedoralabs

Have you been able to verify the updates from dorodnic?

closed 2019-12-01 by dorodnic