realsenseai/librealsense #5217: realsense-viewer in 3d display mode crashes when connected to host using Google remote-desktop

drhalftone, 2019-11-10:
I'm running realsense-viewer on an Up board running Ubuntu 18.04 over a chrome-remote-desktop connection.  Displaying video in 2D mode works fine, but when I switch to 3D display mode, the entire system reboots.

dorodnic, 2019-11-11:
I imagine this has something to do with your OpenGL graphics drivers. One thing that is likely to help is - click on the gears icon in right-top corner, go to settings, navigate to "Performance" tab and disable both "Use GLSL for Rendering" and "Use GLSL for Processing".

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

@drhalftone With the suggestion from dorodnic, did you get it through? Looking forward to your update. Thanks!

closed 2019-12-01 by dorodnic