realsenseai/librealsense #5218: Low fps when aligning depth to color in resolution 1270*720  - rs2 d435

zofnatb, 2019-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer ) | 
| Operating System & Version |   { Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.4.0-141-generic)                                         | 
| Platform                                 | PC nuc  |
| SDK Version                            |  {  2.17 }                          | 
| Language                            |  {C++ on qt }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I've been trying to show the aligned color and depth frames in imshow using different ways but the problem is that I get around 8 fps of images 1270*720. use usb 3.2. 
At lower resolution it 18 fps.
I try to use the issue #2321  but it not help me.
In RealSense-Viewer It seems to have 30 frames per secondץ
there is way to fix that ?

MartyG-RealSense, 2019-11-10:
Anther user who had low FPS performance with imshow and had also read #2321 got better results by building librealsense with CUDA support to offload processing work into the GPU.  

https://github.com/IntelRealSense/librealsense/issues/2376

CUDA support is typically offered in Nvidia graphics chips.  The link below describes how to build with CUDA, and an alternative GPU offload method if you do not have CUDA.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

zofnatb, 2019-11-10:
Tanks, I will  try that and update :)

ev-mp, 2019-11-10:
@zofnatb,  it is highly likely to be the case of CPU being fully utilized. Frame alignment operation is indeed CPU-intensive. You can use `htop` to get actual results.

I'm not sure that you can run CUDA on NUC but on the SDK level you need to verify that the library is compiled with optimizations turned on (`cmake .. -DCMAKE_BUILD_TYPE=release`)

BTW, the viewer does not generate aligned images. To check the performance of `align` class on a specific platform you'd probably need to run `rs-align` and `rs-align-advanced`

zofnatb, 2019-11-10:
@MartyG-RealSense  I don't think I can use CUDA in my NUC. 
@ev-mp  how can I verify that on my SDK level  the library is compiled with optimizations turned on (cmake .. -DCMAKE_BUILD_TYPE=release) ?  Sorry I'm very new to this.
thanks.

ev-mp, 2019-11-10:
@zofnatb, if you're using [Debian packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) then the optimizations are already turned on, so the FPS you get are probably very close to the max that NUC can provide.

In case you build the SDK [from source](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk) then you need to apply the mentioned flag during CMAKE configuration step

You can also experiment with switching to lower resolution/applying decimation filter - this will increase FPS at the expense of depth or color resolutions.

zofnatb, 2019-11-10:
@ev-mp 
Thank you very much for solving this problem.

closed 2019-11-10 by zofnatb