realsenseai/librealsense #5219: Can't connect camera SR300

lxz1104, 2019-11-10:
After upgrade librealsense by apt , it can't detect and  connect the camera SR300 .
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   |  | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-32-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.30.0                      | 
| Language                            |  C++/python                          | 
| Segment			|  others                          | 

### Issue Description
Timestamp: 1573383002959.180176
Severity: Warn
Description: RealSense UDEV-Rules are missing!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger

I have run this commands but it doesn't work.

lxz1104, 2019-11-10:
The terminal output:
```
10/11 18:51:54,133 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:54,412 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:54,612 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:54,812 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:55,013 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:55,226 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:55,439 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:55,633 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
 10/11 18:51:55,853 WARNING [140008846444416] (sr300.cpp:55) try_fetch_usb_device(...) failed.
```

ipa-mah, 2019-11-11:
I have the similar issue when installing librealsense from source. 
The `rs-multicam` only detects D435 camera. SR300 is not detected by the software

lxz1104, 2019-11-11:
Yes,I'm install by source code.
And how can I fix this problem.

ev-mp, 2019-11-13:
@lxz1104 , please be aware that this is a known issue for SR300 with V2.30. Please revert to the previous v2.29.
Duplicate of #5233, #5230

lxz1104, 2019-11-14:
@ev-mp Thanks!I have solved this problem.

closed 2019-11-14 by lxz1104

azeezqad, 2019-11-17:
@lxz1104  How did you solve it ?