realsenseai/librealsense #5220: GET INTRINSIC AND EXTRINSIC PARAMETER ONLY

daothanlong11, 2019-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   |  | 
| Operating System & Version |    Linux (Ubuntu 18.04 LTS)  | 
| Kernel Version (Linux Only)    |  5.0                                      | 
| Platform                                 | PC/ NVIDIA Jetson   |
| SDK Version                            |    2.29.0                           | 
| Language                            |  { C++ }                          | 
| Segment			|   Robots                          | 

### Issue Description
Hello Intel Realsense Dev, I have a D415 camera and i just want to take the intrinsic and extrinsic of this camera, so what file in the librealsense2 source should i include in my code.
I known we can take that parameter by just include all the librealsense library by <librealsense2/rs.hpp>. But i just want to take that 2 matrix only and i dont want to add orther function in my code
Hope I see your answer. Thank for your support

MartyG-RealSense, 2019-11-10:
You can get the intrinsics and extrinsics of a camera with the instructions **get_intrinsics** and **get_extrinsics_to**.    Details of ways to do this are in the link below.

https://github.com/IntelRealSense/librealsense/issues/1706

RealSenseCustomerSupport, 2019-11-25:
----------------------------------------------

Hi @daothanlong11,

Has the link provided by @MartyG-RealSense helped to find an answer to your question?

Thank you!

closed 2019-12-10 by daothanlong11