realsenseai/librealsense #5223: Can't debug the code because of System.Exception: Frame didn't arrived within 5000. UNITY

mintos002, 2019-11-11:
---------------------------------------------------------------------------
| Camera Model                          | D435
| Firmware Version                      | 05.11.15.00 
| Operating System & Version     | Win (10)   
| Platform                                   | PC
| SDK Version                             | 2.29.0 
| Language                                 | unity 
---------------------------------------------------------------------------
I need to align a thermal image to a color image using color and depth. For this I'm using the `RsDevice` code as a base. I've modified the function `WaitForFrames()` like this:
```
    /// <summary>
    /// Worker Thread for multithreaded operations
    /// </summary>
    Align align = new Align(Stream.Color);
    CSRegister.CSRegisterTCD regClass = new CSRegister.CSRegisterTCD(
                    "E:\\_MANTRA\\_imagenes\\patron\\output_imagesx640_80\\spaceing24_extrinsics_640_80.xml", true);
    int counter = 0;
    IntPtr thermo;
    IntPtr thermoOverColor;
    int w, h;
    Texture2D toc_texture;
    public Texture2D tex;

    private void WaitForFrames()
    {
        while (!stopEvent.WaitOne(0))
        {
            using (var frames = m_pipeline.WaitForFrames())
            using (var aligned_frames = align.Process(frames))
                if (aligned_frames.IsComposite)
                {
                    using (var fs = aligned_frames.As<FrameSet>())
                    using (var color = fs.ColorFrame)
                    using (var depth = fs.DepthFrame)
                    {
                        Marshal.FreeHGlobal(thermo);
                        thermo = regClass.readGRAYimage("E:\\path\\t_2019711561978742124.png");

                        if (isRegisterTCD)
                        {
                            Marshal.FreeHGlobal(thermoOverColor);
                            regClass.update(thermo, color.Data, depth.Data, true, true, 0.0f, 1.0f, 0.5f, out w, out h);
                            thermoOverColor = regClass.getThermoOverColor();
                            tex.LoadRawTextureData(thermoOverColor, w * h);
                            tex.Apply();
                            //Debug.Log("REGISTER_" + counter);
                        }

                        counter++;
                        RaiseSampleEvent(aligned_frames);
                    }
                }
        }
    }
```

The thing is that I'm trying to show the `IntPtr thermoOverColor` (a RGB image) to a Unity canvas with `Texture2D` object: `tex.LoadRawTextureData(image, w * h)`. But it shows the following error:

_LoadRawTextureData_ImplPointer can only be called from the main thread.
Constructors and field initializers will be executed from the loading thread when loading a scene.
Don't use this function in the constructor or field initializers, instead move initialization code to the Awake or Start function.
UnityEngine.Texture2D:LoadRawTextureData(IntPtr, Int32)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:208)_

I've tryed to initialize the `Texture2D` tex object into an `Awake()` function but whenever `isRegisterTCD` is true, the frames stops updating.

When I try to debug the code in Visual Studio 2017, it shows the error:

_System.Exception: Frame didn't arrived within 5000 ---> System.Runtime.InteropServices.ExternalException: rs2_pipeline_wait_for_frames(pipe:0000000043923A20)
 --- End of inner exception stack trace ---
 at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (IntPtr pNativeData) [0x00000] in <filename unknown>:0
 at (wrapper managed-to-native) Intel.RealSense.NativeMethods:rs2_pipeline_wait_for_frames (intptr,uint,object&)
 at Intel.RealSense.Pipeline.WaitForFrames (UInt32 timeout_ms) [0x00000] in <filename unknown>:0
 at RsDevice.WaitForFrames () [0x00012] in C:\Users\path\RsDevice.cs:189 happends_

The debug error disappears if I erase the code after the line: using `(var frames = m_pipeline.WaitForFrames())`

Any Idea why I can't debug the code with the example before? or there is any way to increase the waiting frame time?

Also if anyone know how to show an `IntPtr` RGB image in Unity I'll really appreciate that!

Thanks!

ogoshen, 2019-11-12:
The error message is pretty clear, you cannot call `Texture2D.LoadRawTextureData` outside Unity's main thread (like the camera's thread in your case).

You can use the `Intel.RealSense.FrameQueue` for multithreading, see this file:

https://github.com/IntelRealSense/librealsense/blob/2decb32456fc68396da2bc4de25028b1ee2d735f/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsStreamTextureRenderer.cs#L167

mintos002, 2019-11-28:
Thanks! that was it, I've just moved my code inside the unity main thread in RsDevice code.

closed 2019-11-28 by mintos002