realsenseai/librealsense #5224: Can't show public objects in RsDevice inspector. UNITY

mintos002, 2019-11-11:
| Camera Model | D435
| Firmware Version | 05.11.15.00
| Operating System & Version | Win (10)
| Platform | PC
| SDK Version | 2.29.0
| Language | unity
---------------------------------------
Hello there.
I'm trying to modify the RsDevice.cs script to be able to show some public variables that I've added in the code to the Unity inspector. I've been able to use them createing a RsDevice object in code, but I can not see the im the inspector tab.

Does anyone know how to show the variables added in the RsDevice (script) to the Unity Inspector?

Thanks!

wodndb, 2019-11-18:
Check the file "RsDeviceEditor.cs" in RealSesneSDK2.0

mintos002, 2019-11-28:
Thanks! that was it!

closed 2019-11-28 by mintos002