realsenseai/librealsense #5225: D435 orthographic to perspective projection matrix

vladHoloArch, 2019-11-11:
| Camera Model | D435
| Firmware Version | 05.11.15.00
| Operating System & Version | Win (10) / Android 8.1
| Platform | PC / Android
| SDK Version | 2.29.0
| Language | unity

Trying to get the perspective projection matrix derived from d435 rgb module, I've seen the example with ArBackground but it uses the orthographic projection, how to modify this matrix to fit my needs?
  var projectionMatrix = new Matrix4x4
            {
                m00 = intrinsics.fx,
                m11 = -intrinsics.fy,
                m03 = intrinsics.ppx / intrinsics.width,
                m13 = intrinsics.ppy / intrinsics.height,
                m22 = (cam.nearClipPlane + cam.farClipPlane) * 0.5f,
                m23 = cam.nearClipPlane * cam.farClipPlane,
            };

            float r = (float)intrinsics.width / Screen.width;
            projectionMatrix = Matrix4x4.Ortho(0, Screen.width * r, Screen.height * r, 0, cam.nearClipPlane, cam.farClipPlane) * projectionMatrix;

            projectionMatrix.m32 = -1;
            cam.projectionMatrix = projectionMatrix;

ogoshen, 2019-11-12:
The resulting matrix is a perspective matrix, you can see this when running the AR sample.
When using the slider to change the sphere's distance, the sphere will be larger when closer and smaller when far. This won't happen in an orthographic projection.

Can't remember the math from the top of my head, but basically the orthographic matrix is used to transform screen space to normalized coordinates, and the intrinsics used to transform to camera.

In newer Unity version you could use ` Camera.usePhysicalProperties` to configure the camera (sensor size, lens shift, and fov\focal length).

vladHoloArch, 2019-11-12:
@ogoshen thanks, it does makes sense. Besides the projection matrix do I need to simulate the distortion the camera produces?

ogoshen, 2019-11-13:
Yes, though the color stream for D435 has no distortion (intrinsics.coeff[] are all 0)

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

@vladHoloArch,

If the issue is clarified, will close this.

closed 2019-12-25 by RealSenseCustomerSupport