realsenseai/librealsense #5227: OpenNI2 + Realsense D435: FrameSync not supported

kcshum, 2019-11-11:
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.30.0.1174 | 

I encountered an error when using the OpenNI2 wrapper in my application. 

[RS2] ERROR at FILE .../librealsense/wrappers/openni2/src/Rs2Driver.cpp LINE 258 FUNC enableFrameSync
FrameSync not supported

This error is the same as (https://github.com/IntelRealSense/librealsense/issues/5001#issue-503704201) mentioned. Any idea how to fix it?

Looking forward to your reply.

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

Hi @kcshum,

Can you please provide us with your source code that causes to the above-mentioned error. It will make the investigation of the issue more efficient.

Thank you!

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

@kcshum,

Do you still need help with this question?

closed 2020-01-07 by RealSenseCustomerSupport

skychenjian, 2020-05-17:
@kcshum @RealSenseCustomerSupport : 
I faced the same error. Could you please share with me how to fix it ? 
Tks.