realsenseai/librealsense #5228: T265: ERROR: Bulk transfer message 0x25

neilyoung, 2019-11-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.857 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-66 generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.30.0 pre-release                   | 
| Language                            |  C++                          | 
| Segment			|  others                    | 

### Issue Description
This is pretty good reproducible error in the a.m. configuration. Just open either `rs-ar-basic` or `rs-pose-apriltag` sample and close it. Reopen quickly.

The most likely result is:

```
22:34:00.780 [10555] [E] Device-2E70: ERROR: Bulk transfer message 0x25 (DEV_SET_LOW_POWER_MODE) response to host got LIBUSB_ERROR_IO. Host did not return answer
22:34:00.781 [10555] [E] Device-2E70: ERROR LIBUSB_ERROR_NO_DEVICE while control transfer of messageID: 0x11 (CONTROL_GET_INTERFACE_VERSION)
22:34:00.781 [10555] [E] Device-2E70: Error Transferring CONTROL_GET_INTERFACE_VERSION
22:34:00.781 [10555] [E] Device-2E70: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_NO_DEVICE. Bytes requested 6, bytes transferred 0
22:34:00.781 [10555] [E] Device-2E70: USB Error (0x6)
22:34:00.781 [10555] [E] Device-2E70: Error: Failed to get device info (0x6)
22:34:00.781 [10555] [E] Device-2E70: ERROR LIBUSB_ERROR_NO_DEVICE while control transfer of messageID: 0x11 (CONTROL_GET_INTERFACE_VERSION)
22:34:00.781 [10555] [E] Device-2E70: Error Transferring CONTROL_GET_INTERFACE_VERSION
22:34:00.781 [10555] [E] Device-2E70: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_NO_DEVICE. Bytes requested 6, bytes transferred 0
22:34:00.781 [10555] [E] Device-2E70: USB Error (0x6)
22:34:00.781 [10555] [E] Device-2E70: Error: Get Device Info error (0x6)
22:34:00.781 [10555] [E] Device-2E70: Entered state [ERROR_STATE]

```

Subsequent request end up in:

```
22:34:05.470 [10595] [E] TrackingManager: Error while loading image - libusb_bulk_transfer failed. LIBUSB_ERROR_CODE: -7 (LIBUSB_ERROR_TIMEOUT)
```

And mostly there is no other chance as to unplug/replug the device. Also seen with previous version.

bfulkers-i, 2019-12-03:
@neilyoung does this one also happen on PR #5213?

neilyoung, 2019-12-03:
@bfulkers-i Thanks for the follow up. I will set me a reminder to check that over the weekend.

Thanks

neilyoung, 2019-12-05:
@bfulkers-i If not anybody claims the opposite, I would say, this issue is fixed too. USB works as smooth as it is used to be on macOSUSB is as smooth as it is used to be on macOS.

Now I'm still looking for the retained map between several sessions. I think I have read here somewhere, this was subject of a PR, but I can't remember. Maybe I have applied this PR in earlier attempts w/o reminding....https://github.com/IntelRealSense/librealsense/issues/5343

Good job!!!

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @neilyoung

Can we close this thread out or is there anything further needed here?

neilyoung, 2020-01-27:
yes

closed 2020-01-27 by neilyoung