realsenseai/librealsense #5229: modinfo uvcvideo | grep "version:"

netgittemp, 2019-11-11:
I am installing Intel® RealSense™ SDK 2.0.
ubuntu 16.04 
4.15.0-66-generic

`realsense-viewer` is working correctly.
However, when I do `$ modinfo uvcvideo | grep "version:"`, the output cannot see any `realsense` string as you can see the following.
```
version:        1.1.1
srcversion:     79A49299483C7A964CCC049

```
Could you please help me?

likestudy, 2019-11-12:
I meet the same problem with
ubuntu 16.04, 4.10.0-28-generic by following 
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
After the step $  sudo apt-get install librealsense2-utils and connect the realsense sr300, i can open the realsense-viewer but can not get any picture.
Anyone have meet the same problem?

ev-mp, 2019-11-12:
@netgittemp hello, the dkms package for kernel 4.15 is not applicable to versions 4.15.0-58+.(#4669, #4869). The fix is under test. In the meantime the [manual installation guide]((https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)) is available, where you patch the kernel with script:
```sh
./scripts/patch-realsense-ubuntu-lts.sh
```
Note that the script patching does not amend the drivers names. There are ways to verify the patches being applied correctly but in most case following the script log and checking whether the operation finishes successfully is sufficient.

gbr1, 2019-11-12:
Hi @dorodnic, this issue happen also on UP Xtreme with Ubuntu 18.04.3 LTS and kernel 5.0.0-1-generic.
I tried to patch using `patch-realsense-ubuntu-lts.sh` but nothing changed.
Log of installation of `librealsense-dkms`:
```
Setting up librealsense2-udev-rules:amd64 (2.30.0-0~realsense0.1694) ...
Setting up librealsense2-dkms (1.3.6-0ubuntu0) ...
Loading new librealsense2-dkms-1.3.6 DKMS files...
Building for 5.0.0-1-generic
Building initial module for 5.0.0-1-generic
ERROR (dkms apport): kernel package linux-headers-5.0.0-1-generic is not supported
Error! Bad return status for module build on kernel: 5.0.0-1-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.6/build/make.log for more information.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.6: added
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d
```
 
I know that I can compile librealsense from source as explained [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) but are you planning to release an update in the near future?

Thank you in advance!

ev-mp, 2019-11-13:
@gb1, the `librealsense2-dkms` package and `patch-realsense-ubuntu-lts.sh` script are two alternative methods to apply kernel patches. As they affect the same kernel drivers you cannot apply both but have to pick one method - [currently](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages) kernel 5.0 is supported with the later so you need to remove DKMS for that kernel and run the manual script.
Also to clarify - the Debian distributions of the SDK and the kernel patches are not coupled, so it is perfectly fine to to patch kernel with the manual script while using the Debian packages for the library.

Extending DKMS support for LTS kernels 5.0 and 5.3 is scheduled. Follow the [Release Notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes) for updates.

likestudy, 2019-11-13:
@ev-mp I have previously installed kernel patches in two ways. But if I use ‘dkms status ’in the follow-up process, I will delete the extra ones. This should not be the reason why I can't identify the camera.
I saw someone saying that the problem was solved by upgrading the firmware. But I don't know how to check the firmware version of the SR300 and where to upgrade the firmware. Can you give me some help?

ev-mp, 2019-11-13:
@likestudy , I presume you refer to #5230, correct ? I'll follow SR300 identification in that thread.
As for the original question - since you're using kernel 4.15.0-66 then please make sure that the dkms package was not installed for the current kernel ( `dkms status`).

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

@likestudy With the reply of ev-mp, did you get it through now? Looking forward to your update. Thanks!

likestudy, 2019-11-20:
@RealSenseCustomerSupport. Thanks to ev-mp's reply. According to the guidelines of ev-mp, my problem has been solved.

closed 2019-11-21 by netgittemp