realsenseai/librealsense #5230: Realsense camera can not be recognised

likestudy, 2019-11-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (4.10.0-28-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          |                 | 

### Issue Description
The problem I meet is that after follow https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. , the realsense-viewer can not recognize the camera.
**The information by "modinfo uvcvideo | grep "version:" is:**
version: 1.1.2.realsense-1.3.5
srcversion: 35CF6FF13E7340591034B90

**And the output for "usb-devices" is :**
T: Bus=04 Lev=02 Prnt=03 Port=00 Cnt=01 Dev#= 4 Spd=5000 MxCh= 0
D: Ver= 3.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs= 1
P: Vendor=8086 ProdID=0aa5 Rev=31.10
S: Manufacturer=Intel(R) RealSense(TM) Camera SR300
S: Product=Intel(R) RealSense(TM) Camera SR300
C: #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=0mA
I: If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I: If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I: If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I: If#= 3 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I: If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
[output_realsense.txt](https://github.com/IntelRealSense/librealsense/files/3836146/output_realsense.txt)

And i tried :
**1"modinfo uvcvideo | grep ver" :**
version: 1.1.2.realsense-1.3.5
description: USB Video Class driver
srcversion: 35CF6FF13E7340591034B90
vermagic: 4.10.0-28-generic SMP mod_unload
**2\dmesg**
[ 45.950452] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950479] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950489] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950498] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950516] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950525] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.950533] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 45.978527] ACPI Warning: _SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[ 46.111905] nvidia-modeset: Allocated GPU:0 (GPU-5dae5250-fe36-47a7-863f-b9ba3a2f2010) @ PCI:0000:01:00.0
[ 46.112114] nvidia-modeset: WARNING: GPU:0: Correcting number of heads for current head configuration (0x03)
**3\sudo apt-get install librealsense2-dkms**
It shows the librealsense2-dkms is the newest version 1.3.5-0ubuntu1
**4\reboot**
After reboot and "realsense-viewer", the camera still can not be recognized.

- rs-enumerate-devices -c
No device detected. Is it plugged in?

- rs-depth
There are 0 connected RealSense devices.

- rs-capture
RealSense error calling rs2_pipeline_start(pipe:0x1383940):
    No device connected
 Any advices will help me a lot. Thanks.

likestudy, 2019-11-12:
And the information of "dmesg" is :
[ 5389.348780] CPU: 1 PID: 565 Comm: wl_event_handle Tainted: P           OE   4.10.0-28-generic #32~16.04.2-Ubuntu
[ 5389.348782] Hardware name: LENOVO 20192/INVALID, BIOS 6BCN40WW(V2.00) 01/24/2013
[ 5389.348783] Call Trace:
[ 5389.348792]  dump_stack+0x63/0x90
[ 5389.348796]  __warn+0xcb/0xf0
[ 5389.348799]  warn_slowpath_null+0x1d/0x20
[ 5389.348822]  cfg80211_roamed+0x8a/0xa0 [cfg80211]
[ 5389.348887]  wl_notify_roaming_status+0xc5/0x140 [wl]
[ 5389.348943]  wl_event_handler+0x60/0x1e0 [wl]
[ 5389.348950]  kthread+0x109/0x140
[ 5389.349002]  ? wl_notify_scan_status+0x330/0x330 [wl]
[ 5389.349006]  ? kthread_create_on_node+0x60/0x60
[ 5389.349012]  ret_from_fork+0x2c/0x40
[ 5389.349027] ---[ end trace a5aaa8c91cc87a82 ]---
[ 6283.467015] usb 3-3: USB disconnect, device number 5
[ 6283.467020] usb 3-3.2: USB disconnect, device number 6
[ 6283.593499] usb 4-3: USB disconnect, device number 5
[ 6286.398783] xhci_hcd 0000:00:14.0: Cannot set link state.
**[ 6286.398830] usb usb4-port3: cannot disable (err = -32)**
[ 6286.398884] usb 4-3.1: USB disconnect, device number 6
[ 8057.587172] usb 4-3: new SuperSpeed USB device number 7 using xhci_hcd
[ 8057.612428] usb 4-3: New USB device found, idVendor=05e3, idProduct=0616
[ 8057.612433] usb 4-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8057.612435] usb 4-3: Product: USB3.1 Hub
[ 8057.612437] usb 4-3: Manufacturer: GenesysLogic
[ 8057.614987] hub 4-3:1.0: USB hub found
[ 8057.615290] hub 4-3:1.0: 1 port detected
[ 8057.726999] usb 3-3: new high-speed USB device number 7 using xhci_hcd
[ 8057.869513] usb 3-3: New USB device found, idVendor=05e3, idProduct=0610
[ 8057.869517] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8057.869519] usb 3-3: Product: USB2.0 Hub
[ 8057.869521] usb 3-3: Manufacturer: GenesysLogic
[ 8057.870178] hub 3-3:1.0: USB hub found
[ 8057.870434] hub 3-3:1.0: 2 ports detected
[ 8058.719079] usb 3-3.2: new full-speed USB device number 8 using xhci_hcd
[ 8058.978694] usb 3-3.2: New USB device found, idVendor=041e, idProduct=409b
[ 8058.978698] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8058.978701] usb 3-3.2: Product: BlasterX Senz3D VF0810
[ 8058.978703] usb 3-3.2: Manufacturer: Creative Technology Ltd
[ 8058.978705] usb 3-3.2: SerialNumber: 170522183009
[ 8059.124592] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:041E:409B.000C/input/input28
[ 8059.183320] hid-generic 0003:041E:409B.000C: input,hiddev0,hidraw2: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-3.2/input2
[ 8059.863244] usb 4-3.1: new SuperSpeed USB device number 8 using xhci_hcd
[ 8059.884117] usb 4-3.1: New USB device found, idVendor=8086, idProduct=0aa5
[ 8059.884122] usb 4-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8059.884124] usb 4-3.1: Product: Intel(R) RealSense(TM) Camera SR300
[ 8059.884126] usb 4-3.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[ 8059.885633] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[ 8059.887688] uvcvideo 4-3.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 8059.887694] uvcvideo 4-3.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 8059.887698] uvcvideo 4-3.1:1.0: Entity type for entity Extension 3 was not initialized!
[ 8059.887978] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.0/input/input29
[ 8059.888604] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[ 8059.890105] uvcvideo: Unable to create debugfs 4-8 directory.
[ 8059.890253] uvcvideo 4-3.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[ 8059.890256] uvcvideo 4-3.1:1.2: Entity type for entity Camera 5 was not initialized!
[ 8059.890600] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.2/input/input30
[ 8871.228212] usb 3-3: USB disconnect, device number 7
[ 8871.228217] usb 3-3.2: USB disconnect, device number 8
[ 8871.346047] usb 4-3: USB disconnect, device number 7
[ 8871.346082] usb 4-3.1: USB disconnect, device number 8
[ 8889.878586] usb 3-3: new high-speed USB device number 9 using xhci_hcd
[ 8890.021552] usb 3-3: New USB device found, idVendor=05e3, idProduct=0610
[ 8890.021556] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8890.021558] usb 3-3: Product: USB2.0 Hub
[ 8890.021560] usb 3-3: Manufacturer: GenesysLogic
[ 8890.022304] hub 3-3:1.0: USB hub found
[ 8890.022727] hub 3-3:1.0: 2 ports detected
[ 8890.138814] usb 4-3: new SuperSpeed USB device number 9 using xhci_hcd
[... 4 lines cut ...]
[ 8890.164105] hub 4-3:1.0: USB hub found
[ 8890.164460] hub 4-3:1.0: 1 port detected
[ 8891.106654] usb 3-3.2: new full-speed USB device number 10 using xhci_hcd
[ 8891.366193] usb 3-3.2: New USB device found, idVendor=041e, idProduct=409b
[ 8891.366198] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8891.366200] usb 3-3.2: Product: BlasterX Senz3D VF0810
[ 8891.366202] usb 3-3.2: Manufacturer: Creative Technology Ltd
[ 8891.366204] usb 3-3.2: SerialNumber: 170522183009
[ 8891.512048] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:041E:409B.000D/input/input31
[... cut ...]
[ 8892.303726] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.0/input/input32
[ 8892.304549] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[ 8892.306022] uvcvideo: Unable to create debugfs 4-10 directory.
[ 8892.306194] uvcvideo 4-3.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[ 8892.306200] uvcvideo 4-3.1:1.2: Entity type for entity Camera 5 was not initialized!
[ 8892.306516] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.2/input/input33

ev-mp, 2019-11-12:
The highlighted error messages:
>[ 6286.398783] xhci_hcd 0000:00:14.0: Cannot set link state.
[ 6286.398830] usb usb4-port3: cannot disable (err = -32)

can indicate host controller issue. It doesn't necessary mean that the controller fails but you can try switching USB port/cable to mitigate the error. According to the log the UVC driver is properly patched and also loaded correctly, so the issue is probably elsewhere.

Therefore please describe the issue with camera identification/streaming (follow up from the linked issue):
What do you mean by no picture ? Is the camera being recognized by viewer, but you cannot start streaming ? You can attach screenshots as well.
Which sensors did you try to activate and what were the errors you've received ?
Does the device passes enumeration correction ? What does the kernel log show when you plug-in the camera?
Can you collect and provide the realsense application console and kernel logs for
```sh
rs-enumerate-devices -c
rs-depth
rs-capture
```

likestudy, 2019-11-12:
Thanks for your help.

**1.** The problem is that the camera can not be recognized, as below:
[image]

**2.**The sensor I want to activate is the realsense SR300, and the error is this camera can not be recognized.

**3. As I try "rs-enumerate-devices",** I get >>No device detected. Is it plugged in?
And the kernel logs for the camera by 'dmesg' is :
[  145.706081] usb 4-3.4.1: new SuperSpeed USB device number 5 using xhci_hcd
[  145.727059] usb 4-3.4.1: New USB device found, idVendor=8086, idProduct=0aa5
[  145.727063] usb 4-3.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  145.727066] usb 4-3.4.1: Product: Intel(R) RealSense(TM) Camera SR300
[  145.727068] usb 4-3.4.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[  145.728353] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  145.729863] uvcvideo 4-3.4.1:1.0: Entity type for entity Processing 2 was not initialized!
[  145.729870] uvcvideo 4-3.4.1:1.0: Entity type for entity Camera 1 was not initialized!
[  145.729874] uvcvideo 4-3.4.1:1.0: Entity type for entity Extension 3 was not initialized!
[  145.730206] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.4/4-3.4.1/4-3.4.1:1.0/input/input17
[  145.730869] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  145.732121] uvcvideo: Unable to create debugfs 4-5 directory.
[  145.732258] uvcvideo 4-3.4.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[  145.732264] uvcvideo 4-3.4.1:1.2: Entity type for entity Camera 5 was not initialized!
[  145.732506] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.4/4-3.4.1/4-3.4.1:1.2/input/input18
[  507.053616] usb 3-3: USB disconnect, device number 2
[  507.053621] usb 3-3.4: USB disconnect, device number 3
[  507.053624] usb 3-3.4.2: USB disconnect, device number 4
[  507.150961] usb 4-3: USB disconnect, device number 3
[  507.150966] usb 4-3.4: USB disconnect, device number 4
[  507.150968] usb 4-3.4.1: USB disconnect, device number 5
[  514.302677] usb 3-3: new high-speed USB device number 5 using xhci_hcd
[  514.445546] usb 3-3: New USB device found, idVendor=05e3, idProduct=0610
[  514.445550] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  514.445553] usb 3-3: Product: USB2.0 Hub
[  514.445555] usb 3-3: Manufacturer: GenesysLogic
[  514.446305] hub 3-3:1.0: USB hub found
[  514.446733] hub 3-3:1.0: 2 ports detected
[  514.562864] usb 4-3: new SuperSpeed USB device number 6 using xhci_hcd
[  514.585845] usb 4-3: New USB device found, idVendor=05e3, idProduct=0616
[  514.585849] usb 4-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  514.585852] usb 4-3: Product: USB3.1 Hub
[  514.585853] usb 4-3: Manufacturer: GenesysLogic
[  514.588131] hub 4-3:1.0: USB hub found
[  514.588455] hub 4-3:1.0: 1 port detected
[  515.514626] usb 3-3.2: new full-speed USB device number 6 using xhci_hcd
[  515.773930] usb 3-3.2: New USB device found, idVendor=041e, idProduct=409b
[  515.773934] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  515.773937] usb 3-3.2: Product: BlasterX Senz3D VF0810
[  515.773939] usb 3-3.2: Manufacturer: Creative Technology Ltd
[  515.773941] usb 3-3.2: SerialNumber: 170522183009
[  515.918980] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:041E:409B.0004/input/input19
[  515.979152] hid-generic 0003:041E:409B.0004: input,hiddev0,hidraw2: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-3.2/input2
[  516.666723] usb 4-3.1: new SuperSpeed USB device number 7 using xhci_hcd
[  516.687533] usb 4-3.1: New USB device found, idVendor=8086, idProduct=0aa5
[  516.687538] usb 4-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  516.687540] usb 4-3.1: Product: Intel(R) RealSense(TM) Camera SR300
[  516.687542] usb 4-3.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[  516.688986] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  516.690927] uvcvideo 4-3.1:1.0: Entity type for entity Processing 2 was not initialized!
[  516.690931] uvcvideo 4-3.1:1.0: Entity type for entity Camera 1 was not initialized!
[  516.690934] uvcvideo 4-3.1:1.0: Entity type for entity Extension 3 was not initialized!
[  516.691115] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.0/input/input20
[  516.691680] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  516.693163] uvcvideo: Unable to create debugfs 4-7 directory.
[  516.693296] uvcvideo 4-3.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[  516.693299] uvcvideo 4-3.1:1.2: Entity type for entity Camera 5 was not initialized!
[  516.693447] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.2/input/input21
[ 1253.486274] usb 3-3: USB disconnect, device number 5
[ 1253.486280] usb 3-3.2: USB disconnect, device number 6
[ 1253.629869] usb 4-3: USB disconnect, device number 6
[... 21 lines cut ...]
[ 1257.727720] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:041E:409B.0005/input/input22
[... cut ...]
[ 1258.501196] uvcvideo 4-3.1:1.2: Entity type for entity Camera 5 was not initialized!
[ 1258.501399] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.1/4-3.1:1.2/input/input24

**4. The realsense application console and kernel logs are:**
$ rs-enumerate-devices -c    >> No device detected. Is it plugged in?
$ rs-depth     >>There are 0 connected RealSense devices.
$ rs-capture    >> RealSense error calling rs2_pipeline_start(pipe:0x1a7bc20):
    No device connected

And any other information that I can provide to help solve this problem?

ev-mp, 2019-11-13:
@likestudy , based on the info provided we were able to reproduce it locally and unfortunately this is  a software issue introduced in the latest v2.30 version. 
We'll address it in the next release. For now you can rollback to v2.29.
Duplicate of #5233

likestudy, 2019-11-13:
@ev-mp Thanks very much. But i can't find how to rollback to V2.29. As I follow https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md, I get the default version V2.30. And if I build from source code, i also can't find where to choose the version.
Is there any tutorial i can refer to? Thanks again.

likestudy, 2019-11-13:
@ev-mp. I find the link https://github.com/IntelRealSense/librealsense/releases/tag/v2.29.0.
I think I just follow https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md to build from the source code can rollback to v2.29.

likestudy, 2019-11-13:
@ev-mp Thanks a lot for your help. As I rollback to V2.29, the camera can be detected by realsense-viewer. But I use a python script to get video stream, the "No device connected" error occured.
**My code is here:**
_import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
profile = config.resolve(pipeline) # does not start streaming
profile = pipeline.start(config) # Start streaming
depth_sensor = profile.get_device().first_depth_sensor()
motion_range = depth_sensor.get_option(rs.option.motion_range)
val = 0
depth_sensor.set_option(rs.option.motion_range, val)_
**And the error is that:**
    _profile = config.resolve(pipeline) # does not start streaming
RuntimeError: No device connected._

_lsusb | grep 8086_ shows that:
Bus 012 Device 006: ID 8086:0aa5 Intel Corp. 

As I followed [#962](https://github.com/IntelRealSense/librealsense/issues/962) solved this problem by following [Intel® RealSense™ SDK for Linux](https://software.intel.com/sites/products/realsense/sdk/getting_started.html#Setting_Up_Your_C_Development), I get _Fatal error: opencv/cv.h: no such file or directory_ and  _fatal error: librealsense/rs.hpp: no such file or directory_
Is there any other solution?
So sorry to bother you so many times.

fred-och-karlek, 2019-11-13:
hey, I have the same issue. Can I reach you and discuss with you a little bit? @likestudy

ev-mp, 2019-11-13:
@likestudy , please verify that the python package you use is also built with v2.29.
When using Pype - you can get a list of all the previous releases with
```
https://pypi.org/project/pyrealsense2/#history
```
and the install one of the previous versions, like
```
python -m pip install pyrealsense2==2.29.0.1124
```

When building from sources then you can also use git interface (if you using SCM) and checkout tag `v2.29.0`

likestudy, 2019-11-14:
@fred-och-karlek Of course, if I can give you some help.

likestudy, 2019-11-14:
@ev-mp Oh... Thank you very much, I can get the image after I installed the corresponding version of pyrealsense2. After I installed it from source on a computer, I can find realsense information using "modinfo uvcvideo |grep "version" and "dkms status". I can't find it on another computer. I will continue to check the possible reasons.

Jiloc, 2019-11-18:
same problem, SR300 cannot be detected on ubuntu 18.04 after updating to 2.30

likestudy, 2019-11-19:
For my problem, it is caused by the version, which was solved when using v2.29.

closed 2019-11-19 by likestudy