realsenseai/librealsense #5231: Uvcvideo patching issue

markgrts, 2019-11-12:
----------------------------------------------------------------------------------------------------

| Required Info |  
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-32-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0  | 
| Language                            |  C++  | 

----------------------------------------------------------------------------------------------------
I am following the Linux Ubuntu installation guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

When running the `./scripts/patch-realsense-ubuntu-lts.sh` script, it gives me the following error

> Failed to insert the patched module. Operation is aborted, the original module is restored
> Verify that the current kernel version is aligned to the patched module version

I make use of kernel version 5.0.0-32-generic. Is there a patch i can run similar to the kernel 4.16 patch?

ev-mp, 2019-11-12:
@markgrts, the script you're using is the correct one for LTS kernel 5.0.  There could be several reasons for rejecting the patched module, like. Secure Boot or Ubuntu packages inconsistency. 
Please post the kernel log trace when the mentioned error occurs for review (`dmesg`)

markgrts, 2019-11-12:
`dmesg` gives me:

> [  495.611605] usbcore: deregistering interface driver uvcvideo
[  495.931326] media: Linux media interface: v0.10
[  495.931613] Lockdown: modprobe: Loading of unsigned module is restricted; see man kernel_lockdown.7
[  495.960839] videodev: Linux video capture interface: v2.00

ev-mp, 2019-11-12:
@markgrts , googling the error [identifies](https://askubuntu.com/questions/1046269/virtualbox-loading-of-unsigned-modules-is-restricted) this error as one of the symptomps of Secure Boot being enabled.
In order to load patched kernel modules this feature must be disabled.
There are references how to check the Secure Boot  state of the  on Linux - https://www.centos.org/forums/viewtopic.php?t=47224, but to disable it you'll have to enter BIOS settings #2815,#1641

markgrts, 2019-11-12:
Secure Boot was enabled. After disabling it, it worked. Thank you very much!

closed 2019-11-12 by markgrts