realsenseai/librealsense #5232: fatal error: GL/glu.h: No such file or directory     #include <GL/glu.h>

RajkiranVeldur, 2019-11-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } |  435
| Firmware Version                   | (Open RealSense Viewer --> Click info) | NA
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Ubuntu 18.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 5.0.0-32-generic
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | C
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | others

### Issue Description
Issue installing Librealsense **### fatal error: GL/glu.h: No such file or directory     #include <GL/glu.h>**
examples/software-device/CMakeFiles/rs-software-device.dir/build.make:62: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o' failed
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o] Error 1
CMakeFiles/Makefile2:492: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

ev-mp, 2019-11-12:
RajkiranVeldur hello, the include file `<GL/glu.h>` is part of `libglu1-mesa-dev` package  listed in  [Ubuntu18 installation prerequisites](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#ubuntu-build-dependencies) :
```sh
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
```
Please verify that it was properly installed

RajkiranVeldur, 2019-11-12:
Thank you. The issue is resolved. I had to reinstall: sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

ev-mp, 2019-11-13:
Thanks for confirmation, closing it now

closed 2019-11-13 by ev-mp

dmipx, 2023-01-15:
> RajkiranVeldur hello, the include file `<GL/glu.h>` is part of `libglu1-mesa-dev` package listed in [Ubuntu18 installation prerequisites](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#ubuntu-build-dependencies) :
> 
> ```shell
> sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
> ```
> 
> Please verify that it was properly installed

happened to me on ubuntu 20

xyakimo1, 2023-07-04:
For **Fedora**:

`sudo yum install mesa-libGLU-devel`

[Source](https://forums.fedoraforum.org/showthread.php?243265-need-glu-h-GLUT)