realsenseai/librealsense #5233: [Ubuntu18][SR300] is not detected by librealsense

ipa-mah, 2019-11-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |   4.15.0-66-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  v2.30.0                         | 

### Issue Description
I have both SR300 and D435 camera and the librealsense can only detects D435 camera. I checked both build from source code and deb packages. 
Both SR300 and D435 are connected via USB 3.0 port  but only D435 is detected
`modinfo uvcvideo | grep "version:"` outputs:
`version:        1.1.1
srcversion:     4D154F77796C7B15BCB6EFF`
Output from `dmesg` after plugging the camera
[image]
Do you have any idea what is problem?
Thanks

ev-mp, 2019-11-12:
@ipa-mah hello, the kernel log doesn't suggests that anything went wrong; on the contrary, the camera was correctly identified as USB3 device and both the enumeration and loading of the UVC driver completed successfully.

Is it a first time usage ? Do you manage to communicate with this camera on another PC/OS ?
Can you share the output of `rs-enumerate-devices` ?

ipa-mah, 2019-11-12:
@ev-mp 
SR300: 
`rs-enumerate-devices` outputs:
`No device detected. Is it plugged in?`

For 435, it outputs list of modes and device info as usual

ev-mp, 2019-11-13:
@ipa-mah , we managed to reproduce it and can confirm that this is a SR300 bug introduced in the latest v2.30 version. it will be rectified in the next release.
In the meanwhile please use SDK v2.29 or older.
Sorry for the inconvenience

ipa-mah, 2019-11-13:
@ev-mp 
Thank you for your info.
I built the SDK v2.27, v2.28 and had the same issue with SR300. The SDK v2.29 cannot detect camera
`rs-enumerate-devices` outputs:

```
Device info: 
    Name                          : 	Intel RealSense SR300
    Serial Number                 : 	617202004319
    Firmware Version              : 	3.26.1.0
    Recommended Firmware Version  : 	03.26.01.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.2/video4linux/video3
    Debug Op Code                 : 	53
    Product Id                    : 	0AA5
    Camera Locked                 : 	YES
    Product Line                  : 	SR300
    Asic Serial Number            : 	617202004319

Stream Profiles supported by Coded-Light Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  640x480	@ 60Hz	   Y16
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y16
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 10Hz	   Y16
    Infrared 1	  640x480	@ 10Hz	   Y8
    Infrared 1	  640x240	@ 110Hz	   Y16
    Infrared 1	  640x240	@ 110Hz	   Y8
    Infrared 1	  640x240	@ 60Hz	   Y16
    Infrared 1	  640x240	@ 60Hz	   Y8
    Infrared 1	  640x240	@ 30Hz	   Y16
    Infrared 1	  640x240	@ 30Hz	   Y8
    Infrared 1	  640x240	@ 10Hz	   Y16
    Infrared 1	  640x240	@ 10Hz	   Y8
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 10Hz	   Z16
    Depth	  640x240	@ 110Hz	   Z16
    Depth	  640x240	@ 60Hz	   Z16
    Depth	  640x240	@ 30Hz	   Z16
    Depth	  640x240	@ 10Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 10Hz	   RGB8
    Color	  1920x1080	@ 10Hz	   Y16
    Color	  1920x1080	@ 10Hz	   BGRA8
    Color	  1920x1080	@ 10Hz	   RGBA8
    Color	  1920x1080	@ 10Hz	   BGR8
    Color	  1920x1080	@ 10Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
[... 120 lines cut ...]
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 10Hz	   RGB8
    Color	  320x180	@ 10Hz	   Y16
    Color	  320x180	@ 10Hz	   BGRA8
    Color	  320x180	@ 10Hz	   RGBA8
    Color	  320x180	@ 10Hz	   BGR8
    Color	  320x180	@ 10Hz	   YUYV
```
But `realsense-viewer` cannot open `Coded-Light Depth Sensor`, and outputs:

```
13/11 15:35:46,161 ERROR [140353207961344] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 13/11 15:35:46,161 WARNING [140353207961344] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Enable / disable auto-exposure
 13/11 15:35:48,217 ERROR [140353172293376] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device
 13/11 15:35:48,217 ERROR [140353172293376] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device
 13/11 15:35:51,252 WARNING [140353387919232] (backend-v4l2.cpp:1148) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 13/11 15:35:51,252 WARNING [140353387919232] (backend-v4l2.cpp:1148) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 13/11 15:35:51,263 ERROR [140353387919232] (types.h:304) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device 
```
Waiting for next release of the SDK

ev-mp, 2019-11-13:
@ipa-mah,  the camera is properly recognized with the previous version as expected, so this is great.
The error reported at the end is of a different nature:
> 13/11 15:35:48,217 ERROR [140353172293376] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device

The message is fired when the device is forcibly disconnected from/by host. And it often occurs due to device being under-powered. I would recommend to switch USB port and/or get a powered USB3 hub.

ipa-mah, 2019-11-14:
The problem is solved with SDK 2.28.
Thanks

ipa-mah, 2019-12-16:
Hi,
I just want to ask whether the problem of SR300 fixed or not in current `master` branch. Since v2.2.8 version both librealsense and realsense-ros are not working with SR300 (https://github.com/IntelRealSense/realsense-ros/issues/1017)
Thanks

RealSenseCustomerSupport, 2019-12-27:
----------------------------------------------

@ipa-mah SDK v2.31 has fixed the issue. Please try ROS v2.2.11 with SDK v2.31. https://github.com/IntelRealSense/realsense-ros/releases/latest

closed 2019-12-27 by ipa-mah

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@ipa-mah  Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!