realsenseai/librealsense #5237: Error while trying to run the script rs-face-dlib.cpp

RajkiranVeldur, 2019-11-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | NA | 
| Operating System & Version |   Ubuntu 18 | 
| Kernel Version (Linux Only)    |  5.0.0-32-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  NA                          | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue running the script : rs-face-dlib.cpp from librealsense/wrappers/dlib/face

While trying to run the script: gcc rs-face-dlib.cpp

I get the following error:

> /tmp/ccboVyaB.o: In function `main':
> rs-face-dlib.cpp:(.text+0x130d): undefined reference to `dlib::image_window::image_window()'
> rs-face-dlib.cpp:(.text+0x131c): undefined reference to `dlib::base_window::is_closed() const'
> rs-face-dlib.cpp:(.text+0x150f): undefined reference to `dlib::image_window::clear_overlay()'
> rs-face-dlib.cpp:(.text+0x161a): undefined reference to `dlib::image_window::add_overlay(std::vector<dlib::image_display::overlay_line, std::allocator<dlib::image_display::overlay_line> > const&)'
> rs-face-dlib.cpp:(.text+0x16b0): undefined reference to `dlib::image_window::~image_window()'
> rs-face-dlib.cpp:(.text+0x1876): undefined reference to `dlib::image_window::~image_window()'
> /tmp/ccboVyaB.o: In function `__static_initialization_and_destruction_0(int, int)':
> rs-face-dlib.cpp:(.text+0x1bbf): undefined reference to `USER_ERROR__inconsistent_build_configuration__see_dlib_faq_1_'
> rs-face-dlib.cpp:(.text+0x1bcb): undefined reference to `DLIB_VERSION_MISMATCH_CHECK__EXPECTED_VERSION_19_17_0'
> /tmp/ccboVyaB.o: In function `rs2::error::error(rs2_error*)':
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x2a): undefined reference to `rs2_get_error_message'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x76): undefined reference to `rs2_get_failed_function'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x9d): undefined reference to `rs2_get_failed_function'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x100): undefined reference to `rs2_get_failed_args'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x127): undefined reference to `rs2_get_failed_args'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x18a): undefined reference to `rs2_get_librealsense_exception_type'
> rs-face-dlib.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x19f): undefined reference to `rs2_free_error'
> /tmp/ccboVyaB.o: In function `rs2::error::handle(rs2_error*)':
> rs-face-dlib.cpp:(.text._ZN3rs25error6handleEP9rs2_error[_ZN3rs25error6handleEP9rs2_error]+0x22): undefined reference to `rs2_get_librealsense_exception_type'
> /tmp/ccboVyaB.o: In function `rs2::stream_profile::stream_profile(rs2_stream_profile const*)':
> rs-face-dlib.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xc1): undefined reference to `rs2_get_stream_profile_data'
> rs-face-dlib.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xe7): undefined reference to `rs2_is_stream_profile_default'
> /tmp/ccboVyaB.o: In function `rs2::frame::~frame()':
> rs-face-dlib.cpp:(.text._ZN3rs25frameD2Ev[_ZN3rs25frameD5Ev]+0x23): undefined reference to `rs2_release_frame'
> /tmp/ccboVyaB.o: In function `rs2::frame::keep()':
> rs-face-dlib.cpp:(.text._ZN3rs25frame4keepEv[_ZN3rs25frame4keepEv]+0x17): undefined reference to `rs2_keep_frame'
> /tmp/ccboVyaB.o: In function `rs2::frame::get_data() const':
> rs-face-dlib.cpp:(.text._ZNK3rs25frame8get_dataEv[_ZNK3rs25frame8get_dataEv]+0x35): undefined reference to `rs2_get_frame_data'
> /tmp/ccboVyaB.o: In function `rs2::frame::get_profile() const':
> rs-face-dlib.cpp:(.text._ZNK3rs25frame11get_profileEv[_ZNK3rs25frame11get_profileEv]+0x39): undefined reference to `rs2_get_frame_stream_profile'
> /tmp/ccboVyaB.o: In function `rs2::frame::add_ref() const':
> rs-face-dlib.cpp:(.text._ZNK3rs25frame7add_refEv[_ZNK3rs25frame7add_refEv]+0x35): undefined reference to `rs2_frame_add_ref'
> /tmp/ccboVyaB.o: In function `rs2::frame::reset()':
> rs-face-dlib.cpp:(.text._ZN3rs25frame5resetEv[_ZN3rs25frame5resetEv]+0x23): undefined reference to `rs2_release_frame'
> /tmp/ccboVyaB.o: In function `rs2::video_frame::video_frame(rs2::frame const&)':
> rs-face-dlib.cpp:(.text._ZN3rs211video_frameC2ERKNS_5frameE[_ZN3rs211video_frameC5ERKNS_5frameE]+0x6d): undefined reference to `rs2_is_frame_extendable_to'
> /tmp/ccboVyaB.o: In function `rs2::video_frame::get_width() const':
> rs-face-dlib.cpp:(.text._ZNK3rs211video_frame9get_widthEv[_ZNK3rs211video_frame9get_widthEv]+0x3d): undefined reference to `rs2_get_frame_width'
> /tmp/ccboVyaB.o: In function `rs2::video_frame::get_height() const':
> rs-face-dlib.cpp:(.text._ZNK3rs211video_frame10get_heightEv[_ZNK3rs211video_frame10get_heightEv]+0x3d): undefined reference to `rs2_get_frame_height'
> /tmp/ccboVyaB.o: In function `rs2::depth_frame::depth_frame(rs2::frame const&)':
[... 62 lines cut ...]
> rs-face-dlib.cpp:(.text._ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo[_ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo]+0x6a): undefined reference to `dlib::entropy_decoder_kernel_2::entropy_decoder_kernel_2()'
> rs-face-dlib.cpp:(.text._ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo[_ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo]+0x83): undefined reference to `dlib::entropy_decoder_kernel_2::set_stream(std::istream&)'
> rs-face-dlib.cpp:(.text._ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo[_ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decoder_kernel_2ELm200000ELm4EEENS_5crc32EE10decompressERSiRSo]+0xd7): undefined reference to `dlib::entropy_decoder_kernel_2::get_target(unsigned int)'
> rs-face-dlib.cpp:(.text._ZNK4dlib24compress_stream_kernel_1INS_30entropy_encoder_model_kernel_5ILm257ENS_24entropy_encoder_kernel_2ELm200000ELm4EEENS_30entropy_decoder_model_kernel_5ILm257ENS_24entropy_decode [... cut ...]
> rs-face-dlib.cpp:(.text._ZN4dlib13image_display9set_imageI14rs_frame_imageINS_9rgb_pixelELi5EEEEvRKT_[_ZN4dlib13image_display9set_imageI14rs_frame_imageINS_9rgb_pixelELi5EEEEvRKT_]+0x173): undefined reference to `dlib::base_window::invalidate_rectangle(dlib::rectangle const&)'
> rs-face-dlib.cpp:(.text._ZN4dlib13image_display9set_imageI14rs_frame_imageINS_9rgb_pixelELi5EEEEvRKT_[_ZN4dlib13image_display9set_imageI14rs_frame_imageINS_9rgb_pixelELi5EEEEvRKT_]+0x1a3): undefined reference to `dlib::popup_menu_region::disable()'
> collect2: error: ld returned 1 exit status

> 

**I am using dlib-19.17 as suggested**

ev-mp, 2019-11-13:
@RajkiranVeldur, the code you mention is not a stand-alone script but a fully-articulated [Cmake 
 project](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib).
It should be invoked and build from the Librealsense's root using the appropriate flags:
>mkdir build && cd build...
cmake ..   -DCMAKE_BUILD_TYPE=release -D......        ......  **-DBUILD_DLIB_EXAMPLES=true -DDLIB_DIR=<TBD>**

Please follow the [instructions](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib#building-the-dlib-samples-as-part-of-librealsenses-visualstudio-project).

RajkiranVeldur, 2019-11-14:
Hi, @ev-mp I have built it as per the instructions. Can you please help me with the commands to run the project as I'm new to C++ I actually work on Python. Thank you.

RajkiranVeldur, 2019-11-14:
@ev-mp can you please guide me where to place the **shape_predictor_68_face_landmarks.dat** file and how to use it? Thanks for the support.

maloel, 2019-11-14:
Hi @RajkiranVeldur ,
`shape_predictor_68_face_landmarks.dat` needs to be placed in the local directory where the compiled .exe is run from -- usually the same directory that the .exe resides in.

RajkiranVeldur, 2019-11-14:
Hi @maloel I'm now able to run the project. Once again thanks for all the support. Also found that to run this after building you need to use the command **rs-face-dlib**. Closing this issue as it is resolved.

closed 2019-11-14 by RajkiranVeldur