realsenseai/librealsense #5238: Trying to make a new example

collinabidi, 2019-11-13:
|----------------------------------------|------------------------------------------- |
| Camera Model                        | D400                               | 
| Firmware Version                   |  2.29.0                               | 
| Operating System & Version  |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-31                          | 
| Platform                                  |  PC                                       |
| SDK Version                           |  2.0                                | 
| Language                               |  C++/Python                              | 
| Segment			         |  VR/AR                                     | 
### Issue Description
I'm trying to make an application similar to the ```rs-align``` example. My version, called ```rs-align-bag``` takes a .bag file as input and uses the RGB/Depth frames as input instead of a live camera stream.

I run the ```cmake ../ -DBUILD_EXAMPLES=true``` and ```sudo make uninstall && make clean && make && sudo make install``` commands from the Linux manual install guide without errors. The new code package I have is located in it's own directory under librealsense/examples. I have properly modified the corresponding CMake files to recognize this change.

However, after I run the relevant commands to make the examples, there is no ```rs-align-bag``` available on the terminal. When I type ``rs-``` and tab-complete to see relevant commands, all the other ```rs-``` examples are there.

Is there something I'm missing?

neilyoung, 2019-11-13:
I did the same with another example. Basically I just had to create a new directory under examples. I put the new source file and an altered CMakeList.txt, which I too from the original sample.

Then there is a CMakeList.txt in the ./examples folder. You need to `add_subdirectory(your_path)`.

That worked for me.

collinabidi, 2019-11-13:
I did that too, doesn't seem to work for me :/

ev-mp, 2019-11-13:
@collinabidi make sure the new CMake for `rs-align-bag` has [install target](https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/CMakeLists.txt#L14) defined -
>install(TARGETS rs-align-bag RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})

then the binary will be copied to the proper runtime path with `sudo make install`

RealSenseCustomerSupport, 2019-11-25:
----------------------------------------------

Is there any further help needed for this topic?
Thank you.

closed 2019-12-06 by RealSenseCustomerSupport