realsenseai/librealsense #5241: D435i depth map noise problem

dongkaizhou, 2019-11-14:
Hi, 
I am working on a project with D435I, getting the depth map. In general, depth map is normal.
However, when encountering a striped object, the depth map will have a lot of noise. 
Is there any way to reduce noise and avoid this situation?
There are depth map and color map. Thanks!

Haveing striped object (The walls are striped) :
[image]
[image]
[image]

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)          | 
| Platform                                 |  NVIDIA Jetson      |
| SDK Version                            |  legacy / 2.22.0     | 
| Language                            |  {C++/opencv/pcl     | 
| Segment			|  Robot         |

MartyG-RealSense, 2019-11-14:
The presence of fluorescent ceiling strip lights is not ideal as they have the potential to generate more noise on an image than a bulb light due to flickering at frequencies hard to see with the human eye.

Looking at the RGB image, the worst disruption is in the areas where the stripes are most visible due to the illumination from the strip lights.  Looking at the patterns of disruption, it may be due more to the light source than the stripes, since the image is okay in parts where the light on the wall is weaker.

If you are using a 30 FPS speed, could you try 60 FPS please to see if the image improves.  Or if you are using 60, try 30.  This is because disruption from fluorescent light reduces when the FPS is closer to the operating frequency of the light.

dongkaizhou, 2019-11-14:
Hi,
I am using the Intel RealSense Viewer to test 15FPS 30FPS 60FPS, but the noise is the same.
when turning of the light, the noise get normal.
However, my project sense is indoor, there are lights usually.
Are there good ways? Thanks!

Turn on light:
[image]
[image]
[image]
Turn off light:
[image]

MartyG-RealSense, 2019-11-14:
Could you try turning off the Auto Exposure Priority control in the RGB section of the RealSense Viewer options please?   If Auto Exposure is enabled then turning off Auto Exposure Priority should enforce a constant FPS value.

If that does not help, please try turning off Auto Exposure and setting manual exposure and gain values.

RealSenseCustomerSupport, 2019-11-23:
----------------------------------------------

Our RealSense Depth Tuning whitepaperalso contains lots of information for getting the best performance of the D400 series camera.
Please let us know if any further assistance we can provide on this topic. Thank you.

closed 2019-11-27 by RealSenseCustomerSupport