realsenseai/librealsense #5242: Function corresponding to "Heuristic" in the "Occlusion removal" option for RGB 3D pictures in Python

xaritonn, 2019-11-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435
| Firmware Version                   | 05.10.03.00| 
| Operating System & Version |  Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    | 4.15.0-66-generic                                         | 
| Platform                                 | PC  |
| Language                            |  python                          | 

I am making a ".bag" viewer. I need to play a ".bag" file, and be able to switch between a color image and depth map in one window (and at the same time they should be in the same resolution)

I have a frame for depth (848x480) and color (1280x720).
And I need to combine them correctly (fit one image to another)
I tried to align the color in depth, they are combined, but I get a lot of "black spots" and offsets in my color image.

I think this would help me: https://github.com/IntelRealSense/librealsense/issues/2464
Is there any way to implement this in python?

Or is there any other way to correctly combine the two frames?

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi @xaritonn,

You can refer to below post processing doc and apply the filters like hole filtering, etc.
https://dev.intelrealsense.com/docs/depth-post-processing

The Python API as follows.
https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.filter.html

ev-mp, 2019-12-09:
@xaritonn , the use-case you describe is handled by the `alignment` processing block.
You can have it in both depth->color and color->depth ways available based on the application need. It has also been extensively discussed in multiple issues on this forum in the past, so look it up for insights.

The so-called `black-spots` you refer to are actually a desirable feature as during the process the pixel with no match in the target frame are removed to avoid fake results.(#5030

It is not clear what do you mean by  `offset in my color image` but it is possible related to a similar discussion in #5282.

ev-mp, 2019-12-09:
The "Occlusion Removal" filter is implemented as a self-contained processing block but [integrated with  the pointclound](https://github.com/IntelRealSense/librealsense/blob/f361e56f8a2fb5093a1d663293421c92dd432bba/src/proc/pointcloud.h#L45) processing block only.
But since it is open-source there is no reason not to "borrows" and use it in additional use-cases as needed.

xaritonn, 2019-12-20:
Thanks! Your answers have helped a lot.

closed 2019-12-24 by xaritonn