realsenseai/librealsense #5247: Update 2.30 breaks webcam functionality on D435 camera

PhillipVoyle, 2019-11-15:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Ubuntu 16.04.6 LTS | 
| Kernel Version (Linux Only)    | 4.15.0.70-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.30.0-0~realsense0.1693  | 
| Language                            |  | 
| Segment			| others | 

### Issue Description
Chrome fails to receive video from webcam. Chrome shows all 3 devices (depth cameras and RGB camera) and will show a 'Camera is starting' message. I've verified this on more than one site, but namely meet.google.com. Rolling back to 2.29 restores feature

PhillipVoyle, 2019-11-15:
It's worth noting that the realsense-viewer application still worked under these conditions.

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi PhillipVoyle

I am trying to understand the issue you see. Are you saying the the issue only happens on Linux platform? Somehow I do see the Preview in meet.google.com but it is under Window 10 (see the screenshots below). Both D435 and D415 are streaming with RGB cameras. However, it looks I have to close the browser and re-open it when change cameras. Anyway, let me know how to reproduce the issue, then I can look into it.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

Attachment(s):
D415_meet.Google.com.PNG - https://realsensesupport.intel.com/attachments/token/C3XR7NLgYKXwN7WUnSJKW7n1X/?name=D415_meet.Google.com.PNG
D435_meet.Google.com.PNG - https://realsensesupport.intel.com/attachments/token/xxMiuLKi4TqzajNlgSG0FwnI0/?name=D435_meet.Google.com.PNG

PhillipVoyle, 2019-11-24:
Hi there, thanks for coming back. Yes, I noticed the problem on my ubuntu 16 partition, and haven't seen any problems on windows.

PhillipVoyle, 2019-11-24:
```
$ apt list | grep realsense

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

intel-realsense-dfu/xenial 1.1.0 amd64
librealsense2/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-dbg/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-dev/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-dkms/xenial,now 1.3.9-0ubuntu1 all [installed]
librealsense2-gl/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-gl-dbg/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-gl-dev/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-udev-rules/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
librealsense2-utils/xenial,now 2.30.0-0~realsense0.1693 amd64 [installed]
realsense-sdk/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-dbg/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-dev/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-udev-rules/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-utils/xenial 2.7.7-0~realsense0.29 amd64
realsense-uvcvideo/xenial 1.1.0-0ubuntu3 all
```
[image]

PhillipVoyle, 2019-11-24:
Feature is restored under following configuration
```
$ apt list | grep realsense

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

intel-realsense-dfu/xenial 1.1.0 amd64
librealsense2/xenial 2.30.0-0~realsense0.1693 amd64 [upgradable from: 2.29.0-0~realsense0.1579]
librealsense2-dbg/xenial 2.30.0-0~realsense0.1693 amd64 [upgradable from: 2.29.0-0~realsense0.1579]
librealsense2-dev/xenial 2.30.0-0~realsense0.1693 amd64
librealsense2-dkms/xenial 1.3.9-0ubuntu1 all [upgradable from: 1.3.5-0ubuntu1]
librealsense2-gl/xenial 2.30.0-0~realsense0.1693 amd64
librealsense2-gl-dbg/xenial 2.30.0-0~realsense0.1693 amd64
librealsense2-gl-dev/xenial 2.30.0-0~realsense0.1693 amd64
librealsense2-udev-rules/xenial 2.30.0-0~realsense0.1693 amd64 [upgradable from: 2.29.0-0~realsense0.1579]
librealsense2-utils/xenial 2.30.0-0~realsense0.1693 amd64
realsense-sdk/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-dbg/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-dev/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-udev-rules/xenial 2.7.7-0~realsense0.29 amd64
realsense-sdk-utils/xenial 2.7.7-0~realsense0.29 amd64
realsense-uvcvideo/xenial 1.1.0-0ubuntu3 all
```

PhillipVoyle, 2019-11-24:
[image]

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

Hi PhillipVoyle

OK, I see. So it only happens on Ubunto 16.04 with LibRS-2.30.0. but 2.29.0 worked well, right?
Let me submit a ticket to our development team to look into this issue.

Thanks

RealSenseCustomerSupport, 2019-12-10:
----------------------------------------------

Hi PhillipVoyle

I tried D435 with FW5.11.15 on Ubunto 16.04, it seems OK on "meet.google.com". See the screenshot.

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

Attachment(s):
IMG_20191209_181912.jpg - https://realsensesupport.intel.com/attachments/token/XiL77CUjSo5C5Awg86f4Hlp6e/?name=IMG_20191209_181912.jpg

PhillipVoyle, 2019-12-11:
Ok thanks, I have a workaround for my system which is that I don't install the 2.30 update but thought you might like to fix the software at your end

closed 2019-12-11 by PhillipVoyle

PhillipVoyle, 2019-12-16:
I've discovered that removing the dkms package altogether restores the feature also.