realsenseai/librealsense #5249: Detailed kernel version support for librealsense2-dkms

matspetter, 2019-11-15:
Currently it is not possible to see exactly which kernel versions a certain librealsense2-dkms version supports.
For example: EXACTLY which kernel versions are supported by librealsense2-dkms ver 1.3.7 ?
Currently I can only find like 4.15.0-x but it would be good to see somewhere the values of x!!

I see that there are a lot of support issues regarding this (and a lot of trial and error to find a kernel version that works).

dorodnic, 2019-11-17:
Hi @matspetter 
I'll let @ev-mp correct me, but the purpose of the DKMS is to be agnostic to the exact kernel version. Usually, kernel includes do not make breaking changes within single kernel release. 4.15 is the exception, hence the problems. 
We may end up publishing this information (for kernel 4.15.0-x use dkms x.y.z, for kernel 4.15.0-y use dkms a.b.c) but for all other kernel versions we managed to avoid this, and ideally we will find a way here too.

ev-mp, 2019-11-19:
@matspetter hello,
The dkms package are managed per distro, currently Bionic and Xenial as specified in the [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages), and in general those are aligned to [Ubuntu LTS Enablement Roadmap](https://wiki.ubuntu.com/Kernel/LTSEnablementStack) -
Librealsense2-dkms Xenial - handles all the LTS kernels for Ubuntu16 - v4.[4,8,10,13,15].
Librealsense2-dkms Bionic -  currently supports kernels 4.15 and 4.18, with 5.0 and 5.3 scheduled next.
So as @dorodnic noted - there is no specific version, each supported branch should cover all the LTS patch versions released by Ubuntu.

With the kernel branch 4.15 and the unexpected inconsistency in the kernel headers created a break for v4.15.0-58+. (#4669, #4869),
The last  version released (Xenial - 1.3.9, Bionic - 1.3.10) was put under scrutiny for the 4.15 branch.
The installation was tested on kernels 4.15.0-[13,15,23,36,45,55,58,62,66,76], and the functional testing was performed on a subset of those - 4.15.0-[15,20,30,42,52,58,62,66,70].
The branch 4.15 for Bionic was also thoroughly verified with deployment on kernels 4.15.0-[13,15,23,29,33,36,39,42,43,45,46,50,55,58,62,66,70]

matspetter, 2019-11-20:
Ok I see, I was probably unlucky with trying 4.15.70 (or something) just before 1.3.7 was released and I got a bit confused.

closed 2019-11-20 by matspetter