realsenseai/librealsense #5251: D435 Firmware causing USB enumeration problems

boilerbots, 2019-11-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.15.0 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux   | 
| Platform                                 | PC.  |
| SDK Version                            |  NA   | 
| Language                            |  NA                      | 
| Segment			|  Robot     | 

### Issue Description

We are having trouble with the D435 enumerating on the USB bus after reboot. I believe there might be some interaction with the system BIOS during boot.

We have two D435 cameras attached through USB3 ports to hubs and then to the motherboard ports. Sometimes after booting all cameras are enumerated and work, sometimes only 1 and other times none. Additionally sometimes a camera shows up without a serial number when looking in the "/dev/v4l/by-id" directory since that is how we can identify the cameras by their serial numbers.

For example after booting I find

`
ls -l /dev/v4l/by-id/
total 0
lrwxrwxrwx 1 root root 12 Nov 15 10:00 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Nov 15 10:00 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Nov 15 10:01 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Nov 15 10:01 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index1 -> ../../video7
lrwxrwxrwx 1 root root 12 Nov 15 10:01 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index2 -> ../../video4
lrwxrwxrwx 1 root root 12 Nov 15 10:01 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index3 -> ../../video5
`

And after simply rebooting I now see, were one camera is not listing it's serial number and won't work with the ROS node.

`
lrwxrwxrwx 1 root root 13 Nov 15 10:21 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index0 -> ../../video12
lrwxrwxrwx 1 root root 13 Nov 15 10:21 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index1 -> ../../video13
lrwxrwxrwx 1 root root 13 Nov 15 10:22 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index0 -> ../../video10
lrwxrwxrwx 1 root root 13 Nov 15 10:22 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index1 -> ../../video11
lrwxrwxrwx 1 root root 12 Nov 15 10:22 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index2 -> ../../video8
lrwxrwxrwx 1 root root 12 Nov 15 10:22 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index3 -> ../../video9
lrwxrwxrwx 1 root root 12 Nov 15 10:21 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435-video-index0 -> ../../video4
lrwxrwxrwx 1 root root 12 Nov 15 10:21 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Nov 15 10:21 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435-video-index2 -> ../../video2
lrwxrwxrwx 1 root root 12 Nov 15 10:21 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435-video-index3 -> ../../video3
`

When the cameras do not show up properly I can unplug them, plug them in and they always enumerate and function.

After unplugging the camera that is not showing a serial number

`
lrwxrwxrwx 1 root root 13 Nov 15 10:38 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index0 -> ../../video12
lrwxrwxrwx 1 root root 13 Nov 15 10:38 usb-046d_Logitech_Webcam_C930e_E5103D2E-video-index1 -> ../../video13
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index0 -> ../../video6
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index1 -> ../../video7
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index2 -> ../../video8
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523020915-video-index3 -> ../../video9
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index0 -> ../../video4
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index1 -> ../../video5
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index2 -> ../../video2
lrwxrwxrwx 1 root root 12 Nov 15 10:40 usb-Intel_R__RealSense_TM__Depth_Camera_435_Intel_R__RealSense_TM__Depth_Camera_435_920523021709-video-index3 -> ../../video3
`

I have tried to modify various BIOS settings to prevent XHCI handoff, and other legacy USB support but no positive effect was seen.

Typically when a camera isn't listed at all there is also a message in the kernel log that it could not enumerate. When the condition of no serial number in the name, then I don't see any messages in the kernel log.

boilerbots, 2019-11-19:
We conducted a few more tests and have found that plugging the D435 through different hubs and even hubs connected to the primary hubs used in the original ticket the cameras appear to be much more reliable and predictable showing up after reboots. For example Anker 7 port USB 3 hub with camera plugged into first port is very flaky, Plugging another Anker 4 port USB3 hub into the first port and then the same camera made it work 10 out of 10 reboots.

We found other combinations that also worked. 

Perhaps it is a recent kernel regression, I will try some other kernels if I find the time to produce new test images.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Closing this as this issue has been tracked through Zendesk.
Thank you.

closed 2020-01-08 by RealSenseCustomerSupport