realsenseai/librealsense #5252: improve tracking stability and pose accuracy in environments with similar or simple patterns(less feature) for 3d map reconstruction

sadransh, 2019-11-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (any) | 
| Operating System & Version |   {any  | 
| Kernel Version (Linux Only)    |  (4.15)                                         | 
| Platform                                 | PC/linux  |
| SDK Version                            |  { any(2.30) }                          | 
| Language                            |  {any }                          | 
| Segment			|  {Robot and others }                          | 

### Issue Description
I have just started with T265, and I want to do map reconstruction using(both d435i and T265) and probably on a robot with wheel odometry.

However, the environment I am going to work on has a similar pattern and (even mono-color). AFAIK about realsense slam algorithm inside T265, such an environment will affect the robot seriously, and tracking will not work. Is it true? Can adding wheel odometry help the tracking survive such environments? Is there any suggestion to help the system to continue reconstruction? 
What is the source of the problem? Is the weakness of the SLAM inside the camera and using another slam algorithm might solve the issue?  Please clarify this for me to figure out what I have misunderstood.

One more thing to mention, I tried the 3d tracking in front of a wall( like attached image) in realsense-viewer software, the position confidence was mostly on low and sometimes switched to medium for a few moments ( I was moving the camera with hand and slowly  ).

[image]

radfordi, 2019-11-21:
Hi @sadransh,

I would guess that the pattern above wouldn't cause any issues.  I'd even guess that it would help by providing texture our tracking algorithms.    With respect to blank walls, they can indeed be a problem.  That's one of the reasons that T265 ships with very wide angle (~160deg) lenses.  They often allow it to see the corners of a room where a narrow field of view camera would only see the wall.   In the end, it's be best to test is your environment.   Adding wheel odometry can help, but it can take some effort to tune it for your platform.

sadransh, 2019-11-24:
@radfordi  Thank you for your comment, 
I have uploaded a bag file of what's happening. [here](https://1drv.ms/u/s!At8A9wWPQe_ajitFFEuIMHE7yMga?e=knbDLV) is the link. this happens ( ~50 to 60% of the times I'm testing) And might be a fault of my experiment. However, It might help the team. 
some screenshots of it.
[image]
[image]
 
another example, something similar happens in ROS rtabmap : 
[image]
after a few moments, the background of odometry part becomes dark red meaning lost odometry

my own guess is that from the camera perspective the wall is like a blank one. And this seems normal. 

However, I want something to keep tracking in surfaces that have repetitive and similar patterns or simple like the ground, asphalt,... It seems T265 SLAM Algorithm is not designed for such purposes.
Please correct me if I am wrong, Do you have any idea?

RealSenseSupport, 2020-09-21:
Thank you for highlighting the feature of tracking in repetitive patterns or tuning for repetitive patterns. We have moved our focus to our next generation of products and consequently, we will not be addressing this tuning or change in design in the T265.

closed 2020-09-28 by RealSenseSupport