realsenseai/librealsense #5255: VSCode debug configuration?

neilyoung, 2019-11-16:
Hi,
is anybody having here a proper debug configuration for building/debugging one of the C++ samples in Visual Studio Code? I could (at least once) attach VSCode to a running process and also debugging python scripts from VSCode seems to be not a big problem.

Would be great, if one is having such a `launch.json` and would share.

dorodnic, 2019-11-17:
Hi @neilyoung 
My `launch.json` on Mac is:
```js
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            // Resolved by CMake Tools:
            "program": "${command:cmake.launchTargetPath}",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "externalConsole": false,
            "MIMode": "lldb"
        }
    ]
}
```
I'm also using CMake Tools and Native Debug extensions.

neilyoung, 2019-11-17:
@dorodnic Thanks, Sergey.

Not sure if that works on my Linux box. Was installing CMake Tools too, but getting this 
```

[cmake] Configuring done
[cmake] You have changed variables that require your cache to be deleted.
[cmake] Configure will be re-run and you may have to reset some variables.
[cmake] The following variables have changed:
[cmake] CMAKE_C_COMPILER= /usr/bin/x86_64-linux-gnu-gcc-5
[cmake] CMAKE_CXX_COMPILER= /usr/bin/x86_64-linux-gnu-g++-5
[cmake] 
[cmake] The C compiler identification is GNU 5.4.0
[cmake] The CXX compiler identification is GNU 5.4.0
[cmake] Check for working C compiler: /usr/bin/x86_64-linux-gnu-gcc-5
[cmake] Check for working C compiler: /usr/bin/x86_64-linux-gnu-gcc-5 -- works
[cmake] Detecting C compiler ABI info
[cmake] Detecting C compiler ABI info - done
[cmake] Detecting C compile features
[cmake] Detecting C compile features - done
[cmake] Check for working CXX compiler: /usr/bin/x86_64-linux-gnu-g++-5
[cmake] Check for working CXX compiler: /usr/bin/x86_64-linux-gnu-g++-5 -- works
[cmake] Detecting CXX compiler ABI info
[cmake] Detecting CXX compiler ABI info - done
[cmake] Detecting CXX compile features
[cmake] Detecting CXX compile features - done
[cmake] Configuring done
[cmake] CMake Deprecation Warning:
[cmake]   The 'cmake-server(7)' is deprecated.  Please port clients to use the
[cmake]   'cmake-file-api(7)' instead.
[cmake] 
[cmake] 
[cmake] Generating done
[cms-driver] Found no matching code model for the current build type. This shouldn't be possible
[cms-driver] Found no matching code model for the current build type. This shouldn't be possible
[cms-driver] Found no matching code model for the current build type. This shouldn't be possible
[main] =======================================================
[main] No executable target was found to launch. Please check:
[main]  - Have you called add_executable() in your CMake project?
[main]  - Have you executed a successful CMake configure?
[main] No program will be executed
```

neilyoung, 2019-11-17:
With every new start I'm getting asked now, if I want to configure the project. 

The result is always like this:
```

[cmake] Configuring done
[cmake] CMake Deprecation Warning:
[cmake]   The 'cmake-server(7)' is deprecated.  Please port clients to use the
[cmake]   'cmake-file-api(7)' instead.
[cmake] 
[cmake] 
[cmake] Generating done
[cmake] CMake Warning:
[cmake]   Manually-specified variables were not used by the project:
[cmake] 
[cmake]     CMAKE_EXPORT_COMPILE_COMMANDS
[cmake] 
[cmake] 
```

neilyoung, 2019-11-17:
cmake version 3.15.5

neilyoung, 2019-11-17:
I can't build, but I could make gdb run from vscode, changing launch.json like so:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            // Resolved by CMake Tools:
            "program": "/home/decades/Documents/librealsense/build/examples/pose/rs-pose",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "externalConsole": false,
            //"MIMode": "lldb"
        }
    ]
}
```

So, basically provide an absolute path and remove MIMode config

neilyoung, 2019-11-19:
@dorodnic I'm now running the lldb debugger via vscode on macOS. The app is correctly launched, runs and prints out, but none of my breakpoints is hit.

I have enabled logging and as far as I can see one breakpoint is set:

```
--> E (output): {"type":"event","event":"output","body":{"category":"telemetry","output":"VS/Diagnostics/Debugger/Launch","data":{"VS.Diagnostics.Debugger.ImplementationName":"Microsoft.MIDebugEngine","VS.Diagnostics.Debugger.EngineVersion":"14.0.61023.1","VS.Diagnostics.Debugger.HostVersion":"14.0.61023.1","VS.Diagnostics.Debugger.AdapterId":"cppdbg","VS.Diagnostics.Debugger.Launch.Duration":462,"VS.Diagnostics.Debugger.Launch.IsCoreDump":false,"VS.Diagnostics.Debugger.VisualizerFileUsed":false,"VS.Diagnostics.Debugger.SourceFileMappings":0,"VS.Diagnostics.Debugger.MIMode":null}},"seq":2}
--> R (launch-2): {"type":"response","request_seq":2,"success":true,"command":"launch","body":{},"seq":4}
--> E (initialized): {"type":"event","event":"initialized","body":{},"seq":6}
<--   C (setBreakpoints-3): {"command":"setBreakpoints","arguments":{"source":{"name":"rs-pose.cpp","path":"/Users/decades/Documents/librealsense/examples/pose/rs-pose.cpp"},"lines":[286],"breakpoints":[{"line":286}],"sourceModified":false},"type":"request","seq":3}
--> R (setBreakpoints-3): {"type":"response","request_seq":3,"success":true,"command":"setBreakpoints","body":{"breakpoints":[{"id":1,"verified":true,"line":286}]},"seq":9}
--> E (breakpoint): {"type":"event","event":"breakpoint","body":{"reason":"changed","breakpoint":{"id":1,"verified":true,"line":286}},"seq":11}
<--   C (setFunctionBreakpoints-4): {"command":"setFunctionBreakpoints","arguments":{"breakpoints":[]},"type":"request","seq":4}
--> R (setFunctionBreakpoints-4): {"type":"response","request_seq":4,"success":true,"command":"setFunctionBreakpoints","body":{"breakpoints":[]},"seq":14}
<--   C (setExceptionBreakpoints-5): {"command":"setExceptionBreakpoints","arguments":{"filters":[]},"type":"request","seq":5}
--> R (setExceptionBreakpoints-5): {"type":"response","request_seq":5,"success":true,"command":"setExceptionBreakpoints","body":{},"seq":17}
<--   C (configurationDone-6): {"command":"configurationDone","type":"request","seq":6}
--> R (configurationDone-6): {"type":"response","request_seq":6,"success":true,"command":"configurationDone","body":{},"seq":20}
--> E (output): {"type":"event","event":"output","body":{"category":"stdout","output":"Warning: Debuggee TargetArchitecture not detected, assuming x86_64.\n"},"seq":22}
Warning: Debuggee TargetArchitecture not detected, assuming x86_64.
<--   C (threads-7): {"command":"threads","type":"request","seq":7}
--> R (threads-7): {"type":"response","request_seq":7,"success":true,"command":"threads","body":{"threads":[]},"seq":25}
--> E (output): {"type":"event","event":"output","body":{"category":"stdout","output":"=thread-selected,id=\"1\"\n"},"seq":27}
=thread-selected,id="1"
--> E (thread): {"type":"event","event":"thread","body":{"reason":"started","threadId":1},"seq":29}
<--   C (threads-8): {"command":"threads","type":"request","seq":8}

```

Isn't -DCMAKE_BUILD_TYPE=Debug not sufficient to create a gdb debuggable binary?

neilyoung, 2019-11-19:
Additional info: The "-g" flag is set on debug build...

neilyoung, 2019-11-19:
OK, this seems to be a Catalina issue. There is a workaround in form of the CodeLLDB extension, which makes debugging possible again.

https://github.com/microsoft/vscode-cpptools/issues/3829

Can be closed. Solved

closed 2019-11-19 by neilyoung