realsenseai/librealsense #5257: threshold_filter not filtered range

mickeyouyou, 2019-11-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

```c++
  rs2::threshold_filter thr_filter(
      static_cast<float>(FLAGS_point_cloud_min_distance),
      static_cast<float>(FLAGS_point_cloud_max_distance));

  depth_frame = thr_filter.process(depth_frame);
```

mickeyouyou, 2019-11-22:
@dorodnic

dorodnic, 2019-11-25:
Hi @mickeyouyou 
Thank you for the report, sorry for the delay. Your request is clear and makes sense. I'd like @RealSense-Customer-Engineering team to reproduce the issue and if reproduced, it will be addressed as a bug as soon as possible.

mickeyouyou, 2019-11-25:
I am very happy to receive your reply, and I can do any related tests if you need it.

dorodnic, 2019-11-25:
Just to clarify - what I understood from the description is that setting min&max distance via class constructor does not change threshold behavior, while using the options works, is that right?
I did a quick check in `rs-capture`, adding `rs2::threshold_filter tf(0.0f, 0.5f);` after `rs2::rates_printer printer;` and `apply_filter(tf)` after `pipe.wait_for_frames()` and I do see the effect of thresholding (everything above half meter gets filtered out)

mickeyouyou, 2019-11-26:
sorry it's a mistake by myself.  thank you so much.

closed 2019-11-26 by mickeyouyou