realsenseai/librealsense #5258: Building librealsense on macOS Catalina does not work

neilyoung, 2019-11-17:
- Following this tutorial: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md
- First issue: There is no `brew cask install vulkan-sdk`. I know there is a PR pending, the proper command is `brew cask install apenngrace/vulkan/vulkan-sdk`
- CMake runs OK 

```
~/Documents/librealsense/build $ cmake .. -G Xcode -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- The C compiler identification is AppleClang 11.0.0.11000033
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.30.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/local/lib/libusb-1.0.dylib  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Vulkan: /usr/local/lib/libvulkan.dylib  
-- Using Cocoa for window creation
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework   
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Download TM2 firmware 0;"No error" for target-0.2.0.857.mvcmd
-- Download TM2 firmware 0;"No error" for central_app-2.0.19.271.bin
-- Download TM2 firmware 0;"No error" for central_bl-1.0.1.112.bin
-- Fetching recommended firmwares:
-- D4XX_RC_VERSION: 5.11.15.0
-- D4XX_FW_VERSION: 5.11.15.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- Download firmware 0;"No error" for D4XX_FW_Image-5.11.15.0.bin
-- Download firmware 0;"No error" for D4XX_RC_Image-5.11.15.0.bin
-- Download firmware 0;"No error" for SR3XX_FW_Image-3.26.1.0.bin
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   realsense2-gl

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/decades/Documents/librealsense/build

```

- Attempt to make fails: 

```
~/Documents/librealsense/build $ make -j
make: *** No targets specified and no makefile found.  Stop.

```

- Contents of the build folder:

├── CMakeCache.txt
├── CMakeFiles
├── CMakeScripts
├── DerivedData
├── cmake_install.cmake
├── cmake_uninstall.cmake
├── common
├── config
├── examples
├── librealsense2.build
├── librealsense2.xcodeproj
├── realsense2Config.cmake
├── realsense2ConfigVersion.cmake
├── src
├── third-party
├── tools
├── udev-rules.h
└── wrappers

- Also not sure, what the `librealsense2.xcodeproj` is good for. Does not compile in XCode.

dorodnic, 2019-11-18:
Hi @neilyoung 
Have you tried running CMake without `-G Xcode`? Make sure to remove the content of `build` before re-running CMake.

neilyoung, 2019-11-18:
@dorodnic Yepp. Thanks. This worked. Created a PR https://github.com/IntelRealSense/librealsense/pull/5264

I now have a runtime issue, but this is probably something different. The pose sample did run at least once already :)

```
10:47:28.083 [0] [E] Device-1A00: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
10:47:28.311 [0] [E] Device-1A00: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
10:47:28.311 [0] [E] Device-1A00: Entered state [ERROR_STATE]

```

closed 2019-11-18 by neilyoung